pauldemarco/flutter_blue #425: Unable to build iOs

rubberbird, 2019-11-03:
flutter_blue-0.6.3+1/

`fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found
    #import <flutter_blue/FlutterBluePlugin.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    note: Using new build systemnote: Planning buildnote: Constructing build description`

Ran flutter clean, flutter build. Same issue. Used to work on 6.1

DaYuClo, 2019-11-05:
I am getting the same error and is unable to run or build iOS since updating to mac 10.15.
I have tried 0.6.1, 0.6.2, 0.6.3, and 0.6.3+1 without success.

Flutter 1.9.0
XCode 11.2
Mac 10.15

ekuleshov, 2019-11-07:
Same issue here, but I'm on osx 10.13.6 and xcode 10.1. Any advice?

abjox, 2019-11-10:
I have the same issue. 

Flutter  1.10.15
XCode 11.2
Mac 10.14.5

xxzj990, 2019-11-12:
me too.

Flutter 1.9.1+hotfix.5
Xcode  11.0
Mac     10.15

Nailer, 2019-11-14:
+1

Flutter 1.9.1+hotfix.5
XCode 11.2.1
Mac 10.14.6

ekuleshov, 2019-11-15:
Adding this as a reference, just in case

https://github.com/flutter/flutter/issues/16049

ZmanElete, 2019-11-15:
I ran into this issue after updating cocoapods to 1.8.4, I scaled it back to 1.8.0 and got the same error then jumped back to 1.7.5 and it compiled fine again.

furkanvatandas, 2019-11-19:
+1 

`fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found #import <flutter_blue/FlutterBluePlugin.h> ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 1 error generated.`

BrunoEleodoro, 2019-11-20:
Same here :/ 

↳
    4 warnings generated.
    ios/Runner/GeneratedPluginRegistrant.m:6:9: fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found
    #import <flutter_blue/FlutterBluePlugin.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    note: Using new build systemnote: Planning buildnote: Constructing build description

BrunoEleodoro, 2019-11-20:
I did this guys , worked for me!

https://github.com/pauldemarco/flutter_blue/issues/406#issuecomment-544530920

ekuleshov, 2019-11-20:
That only works if you not using other swift-based plugins in your app. With swift plugins you cant remove that option.

BrunoEleodoro, 2019-11-20:
Oh sorry @ekuleshov , didn't notice that... let's wait for other solutions.

xxzj990, 2019-11-22:
@all
worked for me,thanks @julienzarka:

> flutter_blue:
>  git:
>    url: https://github.com/julienzarka/flutter_blue.git

DaYuClo, 2019-11-26:
> I ran into this issue after updating cocoapods to 1.8.4, I scaled it back to 1.8.0 and got the same error then jumped back to 1.7.5 and it compiled fine again. I want to emphasize I jumped versions, I did not hit every one in between 1.8.0 and 1.7.5

Downgrading cocoapods to 1.7.5 fixes my problem.  Thank you.

mac 10.15.1
flutter_blue: ^0.6.0+3
Dart 2.7.0
Flutter 1.12.13
Xcode 11.2.1

pauldemarco, 2020-04-02:
Hi all,

I committed a change yesterday that attempts to fix this problem.  Please test the master branch of flutter blue and let me know your results.

Thanks.

julienzarka, 2020-04-03:
Hi Paul,

it works for me ! Thanks for the hard work.

Kind regards,

danfleck, 2020-04-03:
I still see the problem. I updated my podspec.yaml to:
  flutter_blue: ^0.7.0

Then upgraded the plugins. Did a "flutter clean" 

When I build IOS I still get an error on flutter_blue:

File not found: #import "gen/Flutterblue.pbobjc.h"

I attached an XCode screenshot if it helps.
[image]

danfleck, 2020-04-06:
Tested the new version (0.7.1) and it works now building for iOS using Android Studio.  So, I'm good. However it does not build using XCode. The error I get (only in XCode) is:

`error: open /Users/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1/ios/gen/Flutterblue.pbobjc.h: No such file or directory (in target 'flutter_blue' from project 'Pods')
`
It seems like it doesn't find the username when building in XCode for some reason. Full error in screenshot attached.

Thanks for all your work on this Paul! If this is a different issue and should be opened in its own issue, let me know!

[image]

janniklind, 2022-03-10:
I just experienced the above mentioned problem with error:

`error: open /Users/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1/ios/gen/Flutterblue.pbobjc.h: No such file or directory (in target 'flutter_blue' from project 'Pods')`

I noticed that it was looking in `/Users/.pub-cache/....`, but my .pub-cache folder has always been located in `/Users/jlind/.pub-cache/....`

I need to find a better fix, but I was lazy and tried to create a symlink:

`sudo ln -s /Users/jlind/.pub-cache /Users/.pub-cache`
It works now :)

boskokg, 2022-03-10:
@janniklind use flutter_blue_plus

Forsenen, 2022-04-09:
> I just experienced the above mentioned problem with error:
>
> `error: open /Users/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1/ios/gen/Flutterblue.pbobjc.h: No such file or directory (in [...]

can you explain more. I am having the same error

boskokg, 2022-04-09:
go to flutter_blue_plus