pauldemarco/flutter_blue #426: connected  ' discoverServices '  is't get

LiangLuDev, 2019-11-04:
Hi @pauldemarco 
```
List<BluetoothDevice> devices = await FlutterBlue.instance.connectedDevices;
List<BluetoothService> services= await devices[0].discoverServices();
print('$services');
```
Android  error  :
java.lang.IllegalArgumentException: Unsupported value: java.lang.Exception: no instance of BluetoothGatt, have you connected first?

iOS  error:
not print log and services

TKS

ride4sun, 2020-04-09:
@LiangLuDev yes the device has to be connected before you can call dicoverServices. The new version will throw an exception when you don't. Can this be closed?

LiangLuDev, 2020-04-09:
@ride4sun current connected.
  iOS : Although the settings show that it is connected, it still needs to be connected in the app

ride4sun, 2020-04-09:
Can describe that behavior a bit better? - Possibly with some code.

woopluz, 2020-10-20:
same issuses.

EricTURATSINZE, 2024-01-06:
Facing same issue