pauldemarco/flutter_blue #432: Error with reading a characteristics: PlatformException(read_characteristic_error...

marileen-jansen, 2019-11-08:
Hi. I get an exeption when trying to read a characteristic ( c.read() )

Here is the exeption: 
_Exception has occurred.
PlatformException (PlatformException(read_characteristic_error, unknown reason, may occur if readCharacteristic was called before last read finished., null))_

Can you please help me with this?

flonguini, 2019-11-13:
@marileen-jansen Did you solve this problem?

marileen-jansen, 2019-11-19:
> @marileen-jansen Did you solve this problem?

No I have not.

chaochaox1990, 2019-11-22:
add one

LeviDahl, 2020-01-16:
I realized I had a characteristic that did not have a read property in my list of characteristics.

I added this if statement and everything seemed to work fine.

```
if (c.properties.read)
        {
          List<int> value = await c.read();
          print(value);
        }
```

Mozyra, 2020-02-09:
Still same problem.

aytunch, 2020-02-09:
This library is unfortunately have not been maintained lately.
Have you guys tried https://github.com/Polidea/FlutterBleLib?
It looks promising. I will move there I guess.

pauldemarco, 2020-02-09:
@aytunch The plugin at FlutterBleLib is maintained by the Polidea folks, and utilizes their tried and tested platform libs for Android and iOS.  I'm sure you'll be in good hands.

FlutterBlue is still under development and I'll be disclosing project plans for 2020 in the coming weeks.

Thanks.

Mozyra, 2020-02-10:
forEach doesn’t wait for callback return, it just launches them and returns. If you are going to perform async operations inside forEach, you have to use «for in» instead.

Don’t:
services.forEach((service) async {
    // do something with service
});
Do:
for (BluetoothService service in services) {
   // do something with service
}

CrazyAndy, 2020-03-15:
you should listen to it.

for(BluetoothService service in services){
      print("service id is : ${service.uuid}");
      if (_matchSmartLockServiceId(service.uuid.toString())) {
        var characteristics = service.characteristics;
        for (BluetoothCharacteristic c in characteristics) {
          print("### ${c.uuid}");
          String c_id = c.uuid.toString();
          if (_getBlueToothFunctionCode(c_id) == "****"){
            await c.setNotifyValue(true);
            c.value.listen((value) {
              _readInfoFromDevice(value);
            });
          }else if (_getBlueToothFunctionCode(c_id) == "****"){
            _sendMsg(c);
          }
        }
      }
 _readInfoFromDevice(List<int> values) async{
    if(values.length>0){
      print("value: ${values}");
    }
  }
  _sendMsg(BluetoothCharacteristic c) async{
    //do something
  }    
bool _matchSmartLockServiceId(String serviceId) {
//do your logic
    if (serviceId.isEmpty) return false;
    if (serviceId.substring(0, 4) == "****") return true;
    return false;
  }

jiyarong, 2020-07-09:
+ 1

realrk69, 2020-12-05:
Confirmed: This happens on iOS only for me now. Android is working 100%. Shifted to flutter_reactive_ble and it works perfectly there.

qurm, 2021-01-09:
Note that this only works when the characteristic support Notifications - not all will do.

so check for `c.isNotifying` and then either read, or listen for a notification

> you should listen to it.
> 
> if (_getBlueToothFunctionCode(c_id) == "****"){ await c.setNotifyValue(true); c.value.listen((value) { _readInfoFromDevice(value);

jjanixx, 2021-01-12:
I get the above mentioned error even if I try to run the Flutter Blue example code. The moment I press the sync and try enabling notifications for a characteristic, the error is thrown.

AryanSethi, 2021-05-07:
@qurm I'm new to this and I'm facing the same issue. Can you please tell me what the ```read()``` and ```listen()``` functions are doing and why I should be using one over the other.

yuriboeira11tx, 2021-05-17:
Any suggestions? Same problem in version 0.8.0!

cthurston, 2021-05-20:
We just hit this bug as well. The only information that I can give that might be helpful:
When I place a breakpoint on the read() call it will successfully complete the read. This suggests that if you attempt to do two different actions, even on different characteristics that it will throw. For example if I'm subscribed(notify) on one characteristic and my code does something to prompt a notification and at the same time requests a read, the BLE request fails.

Outside of some lib level queue, retry or wait to guarantee that you don't do two things at once, I'm not sure how to fix/workaround.

ios9developer, 2021-09-28:
Hello everyone, the problem is finally solved, there must be a time interval between two data readings. Thanks to the author, I hope to update it quickly to solve the problem that the Android device cannot be disconnected for a long time after a successful connection.

大家好，问题最终解决了，两次数据读取必须要有一个时间间隔。感谢作者希望能快点更新，解决安卓设备连接成功后，长时间无法断开问题。

```
    if (charUuid == "FirstCharId") {
      await characteristic.read();
    } else if (charUuid == "SecondCharId") {
      Future.delayed(Duration(milliseconds: 500), () async {
        await characteristic.read();
      });
    }
```

Mitchlol, 2021-12-19:
Was running into this issue, turned out to be 2 problems for me.

1) I did not have a READ property set on my characteristic. I had NOTIFY, but not read. This meant I could get partial data by listening, but could not read the full message.
2) I could not read while sending a message, so I had to wait for the send to complete before I could read. Even though I was sending a message to a different characteristic.

Hope this helps someone, BLE is hard 😅

hammvdh, 2022-03-09:
Still couldnt find a resolve for this, please let us know if there is any update on this for a fix

Twocram, 2022-04-13:
Hello world! I solved the problem this way:

`
  getDeviceInfo() async {
    List<BluetoothService> services = await widget.device.discoverServices();
    for (final BluetoothService service in services) {
      var characteristics = service.characteristics;
      for (final BluetoothCharacteristic c in characteristics) {
        if (c.properties.read) {
          List<int> value = await c.read();
          if (value.isNotEmpty) {
            try {
              final String decodedBytes = utf8Decoder.convert(value);
              decodedValues.add(decodedBytes);
            } catch (e) {
              print("ERROR DECODING ${e}");
            }
          } else {
            print("NO DATA");
          }
        }
      }
    }
  }`

Twocram, 2022-04-13:
Hello world. I solved the problem in the following way:
```
  getDeviceInfo() async {
    List<BluetoothService> services = await widget.device.discoverServices();
    for (final BluetoothService service in services) {
      var characteristics = service.characteristics;
      for (final BluetoothCharacteristic c in characteristics) {
        if (c.properties.read) {
          List<int> value = await c.read();
          if (value.isNotEmpty) {
            try {
              final String decodedBytes = utf8Decoder.convert(value);
              decodedValues.add(decodedBytes);
            } catch (e) {
              print("ERROR DECODING ${e}");
            }
          } else {
            print("NO DATA");
          }
        }
      }
    }
  }
```

azizur-rahaman, 2023-07-01:
> 

Thanks man it's work for me.

shubhamsinghshubham777, 2023-07-05:
Other given answers helped me as well 👍🏻 
I'm just trying to write what I observed.

We cannot read values from a characteristic after we have started observing them, and if we just start observing them (i.e. by setting `setNotifyValue(true)` and then listening to the value stream) then it won't provide any initial value, it will only start emitting if the given characteristic's value has been changed at least once.

Therefore, to tackle this issue, I first read the characteristic and cached its response in a variable, and only then started observing it.

### Code example
```dart
Future<Stream<Color>> ledColor({required String initialHexColor}) async {
    final ledColorCharacteristic = characteristic(BLEConstants.ledInfoCharacteristic);

    // This can be tricky (i.e. always remember that if you want to both read
    // and observe a characteristic, then always complete the read operation
    // first and only then start observing its values, otherwise the app will
    // throw a read_characteristic_error. That's why, even though we'll not use
    // this read value instantly, we're caching it before we have to
    // setNotifyValue(true).
    final values = await ledColorCharacteristic.read();

    if (!ledColorCharacteristic.isNotifying) {
      await ledColorCharacteristic.setNotifyValue(true);
    }

    Stream<Color> getColorStream() async* {
      yield _colorFromValues(values);
      yield* ledColorCharacteristic.value.map(_colorFromValues);
    }

    return getColorStream();
  }
```