pauldemarco/flutter_blue #433: Android targeting sdk version 29 not working with version 0.6.3+1

tomcatvr, 2019-11-11:
Hi,
if I switch to targetSdkVersion 29, the example starts but it does not found any devices.

Step to reproduce:
- use Flutter 1.9.1+hotfix.6 • channel stable
- download example
- add implementation "androidx.core:core:1.1.0" in build.gradle so we can use flutter stable channel
- switch to targetSdkVersion 29.
Same behaviour if I use flutter master channel

rjcasson, 2019-11-11:
Do you happen to using a device running Android 10? I ran into this issue a few weeks back and made a pull request that fixed it on my end: #383

tomcatvr, 2019-11-12:
@rjcasson Yes, thanks! I'm running Android 10.
Waiting for the update I'll compile with targetSdkVersion 28.

closed 2019-11-12 by tomcatvr