pauldemarco/flutter_blue #434: Install Does Not Work

jonhunterking, 2019-11-11:
Hello,

I have just tried to follow the flutter_blue instructions for install. I have done the following:

1. Added "flutter_blue: ^0.6.3+1 to my pubspec.yaml file under my dependencies. That section is now as follows:

dependencies:
  flutter:
    sdk: flutter
  cupertino_icons: ^0.1.2
  flutter_blue: ^0.6.3+1

2. I ran in my IDE and in my command window "flutter pub get" and it reported no errors 
3. I then added import 'package:flutter_blue/flutter_blue.dart'; to my main.dart file and it was underlined red. When I moused over it there was an error like "flutter blue target of uri does not exist"

I then restarted my IDE. The red underline was gone. However, when I tried to run my program I got the following error:

_"C:\Users\jonathan.king\App Dev\oto_app_0\android\app\src\debug\AndroidManifest.xml Error:
	uses-sdk:minSdkVersion 16 cannot be smaller than version 19 declared in library [:flutter_blue] C:\Users\jonathan.king\App Dev\oto_app_0\build\flutter_blue\intermediates\merged_manifests\debug\processDebugManifest\merged\AndroidManifest.xml as the library might be using APIs not available in 16
	Suggestion: use a compatible library with a minSdk of at most 16,
		or increase this project's minSdk version to at least 19,
		or use tools:overrideLibrary="com.pauldemarco.flutter_blue" to force usage (may lead to runtime failures)

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : uses-sdk:minSdkVersion 16 cannot be smaller than version 19 declared in library [:flutter_blue] C:\Users\jonathan.king\App Dev\oto_app_0\build\flutter_blue\intermediates\merged_manifests\debug\processDebugManifest\merged\AndroidManifest.xml as the library might be using APIs not available in 16
  	Suggestion: use a compatible library with a minSdk of at most 16,
  		or increase this project's minSdk version to at least 19,
  		or use tools:overrideLibrary="com.pauldemarco.flutter_blue" to force usage (may lead to runtime failures)

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 29s

Finished with error: Gradle task assembleDebug failed with exit code 1"_

tomcatvr, 2019-11-11:
Hi, you have to change to minSdkVersion 19 in your build.gardle configuration
Follow https://github.com/pauldemarco/flutter_blue/issues/415#issuecomment-548186492
If you want to use flutter stable channel you also have to update you build.gradle file of the flutterblue plugin as here

dependencies {
    implementation 'com.google.protobuf:protobuf-lite:3.0.1'
    implementation "androidx.core:core:1.1.0"
}

jonhunterking, 2019-11-11:
@tomcatvr I posted in #415. Following the recommendation from @pauldemarco broke my project so I am trying to find a way out xD I posted my issue there.

tomcatvr, 2019-11-12:
That's strange. Have you tried to compile just only the example?
You only have to change the minSdkVersion and add implementation "androidx.core:core:1.1.0". 
Then update your gradle.plugin and should work.

jonhunterking, 2019-11-12:
Right now I have my sdk version set to >=2.1.0 < 3.0.0. What should I change it too?

Also I missed your other step about changing the dependencies in the build.gradle file. It is currently:
    dependencies {
        classpath 'com.android.tools.build:gradle:3.2.1'
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlin_version"
    }

But I will change it to:
    dependencies {
        classpath 'com.android.tools.build:gradle:3.2.1'
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlin_version"
        implementation 'com.google.protobuf:protobuf-lite:3.0.1'
        implementation "androidx.core:core:1.1.0"
    }

tomcatvr, 2019-11-12:
I think you are confusing between variuos gradle file
I you use AndroidStudio, you can see under gradle
[image]

build.gradle(Module:flutter_blue) has this dependencies
dependencies {
    implementation 'com.google.protobuf:protobuf-lite:3.0.1'
    implementation "androidx.core:core:1.1.0"
}

while build.gradle(Module:app) in my case has
dependencies {
    testImplementation 'junit:junit:4.12'
    androidTestImplementation 'androidx.test:runner:1.1.1'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.1.1'
}

and build.gradle(Project:android) has
dependencies {
     classpath 'com.android.tools.build:gradle:3.5.2'
 }

jonhunterking, 2019-11-12:
I reinstalled my android studio and things seem to be working now. I am not using the package now so my "import 'package:flutter_blue/flutter_blue.dart';" is just underlined grey as "unused import" which seems right. No issues with debugging it as such. I think I may have missed a step in my install? I will close this issue as I believe it is resolved within the context of "install" being an issue.

closed 2019-11-12 by jonhunterking