pauldemarco/flutter_blue #437: Scan filter withService does return result

saulopef, 2019-11-12:
I have a fisical device, with an custom service:
`  final String myService = "4fafc201-1fb5-459e-8fcc-c5c9c331914b";`
I tryed scan with filter "withServices" with:
`flutterBlue.startScan(
timeout: Duration(seconds: 4),
withServices: [Guid(myService)]
);`
and there is no result, but when I coment the line "withServices: [Guid(myService)]", the device is conected and the services is descoveres, including the myService corresponding service.

I have tested this function with a HR frequenci device with service 180D, and it's works fine, I beliave the problem is the custom service, what I'm doing wrong?

johnnyoin, 2019-11-20:
Does your device actually advertise the service UUID? Otherwise it can't be filtered that way.

rknell, 2019-11-21:
I can confirm this does not appear to be working.

I have a device that has the service UUID listed in the `advertisementData` of the ScanResult. If I run 
`scanResult.advertisementData.serviceUuids.contains("0000fff0-0000-1000-8000-00805f9b34fb") //true`

it returns true. However if I do

`flutterBlueInstance.scan(
            timeout: Duration(seconds: 5),
            withServices: [Guid("0000fff0-0000-1000-8000-00805f9b34fb")]
          )`

it returns no items.
----

So what does this mean? Well its broken, but a workaround is to just filter devices on the ScanResult before doing any work.

johnnyoin, 2019-11-21:
It works very well for me. Maybe try uppercased?

saulopef, 2019-12-11:
yes, the error was in the adversmentData of the device, not in code, tks for your time.

closed 2019-12-11 by saulopef