pauldemarco/flutter_blue #438: Read characteristics throws exception

flonguini, 2019-11-13:
Hello,

I'm trying to read the characteristics but the method read() from BluetoothCharacteristic always throw an exception.

> Exception has occurred.
PlatformException (PlatformException(read_characteristic_error, unknown reason, may occur if readCharacteristic was called before last read finished., null))

My code is:

```
 getMsg1() async {
    var characteristics = _sierraCustomService.characteristics;
    for (var characteristic in characteristics){
      List<int> msg = await characteristic.read();
      print('Charac: $msg');
    }
}
```

 _sierraCustomService.characteristics create a list with 6 items in characteristics. The debug output is:

```

D/FlutterBluePlugin(26917): [onCharacteristicChanged] uuid: 06d1e5e7-79ad-4a71-8faa-373789f7d93c
D/FlutterBluePlugin(26917): [onCharacteristicChanged] uuid: 06d1e5e7-79ad-4a71-8faa-373789f7d93c
D/FlutterBluePlugin(26917): [onCharacteristicChanged] uuid: 06d1e5e7-79ad-4a71-8faa-373789f7d93c
D/FlutterBluePlugin(26917): [onCharacteristicChanged] uuid: 06d1e5e7-79ad-4a71-8faa-373789f7d93c
D/FlutterBluePlugin(26917): [onCharacteristicChanged] uuid: 06d1e5e7-79ad-4a71-8faa-373789f7d93c
D/FlutterBluePlugin(26917): [onCharacteristicChanged] uuid: 06d1e5e7-79ad-4a71-8faa-373789f7d93c
I/flutter (26917): remoteId: 20:FA:BB:0B:CE:19 characteristicUuid: 06d1e5e7-79ad-4a71-8faa-373789f7d93c serviceUuid: bc2f4cc6-aaef-4351-9034-d66268e328f0

```

I know that the last row is from the read method.

```
FlutterBlue.instance._log(LogLevel.info,
        'remoteId: ${deviceId.toString()} characteristicUuid: ${uuid.toString()} serviceUuid: ${serviceUuid.toString()}');

```

Am I doing something wrong?

LeviDahl, 2020-01-16:
I had this issue because one of my characteristics didn't have the read property. This fixed it.
```
if (c.properties.read)
        {
          List<int> value = await c.read();
          print(value);
        }
```

pauldemarco, 2020-04-02:
@flonguini have you attempted @LeviDahl’a fix?
Any further updates since your posting in November (sorry for the late response)?

teekoksiang, 2020-06-19:
I also experienced this issue, but after debugging for a while, I found out that I called the function to read the characteristic twice. So, I fixed it by calling the function once only.

aytunch, 2020-06-23:
@teekoksiang I see. But still is it the expected behavior for the library to throw an exception when a specific characteristic is read twice consecutively? I think it should allow it none the less.

teekoksiang, 2020-06-24:
@aytunch I guess so, as the error message stated that please wait for the previous characteristic read to finish, and I called the read function twice for the same characteristic.

novas1r1, 2020-06-28:
so how can I then wait for it to finish? Or detect when its finished? Running into this problem when reading the characteristic more than once in a short amount of time.