pauldemarco/flutter_blue #441: Error Running Flutter Blue Example

jonhunterking, 2019-11-13:
hello,

I am currently trying to run the flutter blue example. I want to get it running then publish a test app to see if I can connect to some devices. However, when I try and run the program I get the following error:

"C:\flutter_blue-master\flutter_blue-master\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:43: error: package androidx.core.app does not exist
import androidx.core.app.ActivityCompat;
                        ^
C:\flutter_blue-master\flutter_blue-master\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:44: error: package androidx.core.content does not exist
import androidx.core.content.ContextCompat;
                            ^
C:\flutter_blue-master\flutter_blue-master\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:150: error: cannot find symbol
                if (ContextCompat.checkSelfPermission(activity, Manifest.permission.ACCESS_COARSE_LOCATION)
                    ^
  symbol:   variable ContextCompat
  location: class FlutterBluePlugin
C:\flutter_blue-master\flutter_blue-master\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:152: error: cannot find symbol
                    ActivityCompat.requestPermissions(
                    ^
  symbol:   variable ActivityCompat
  location: class FlutterBluePlugin
Note: C:\flutter_blue-master\flutter_blue-master\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\flutter_blue-master\flutter_blue-master\example\build\flutter_blue\generated\source\proto\debug\javalite\com\pauldemarco\flutter_blue\Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
4 errors
*******************************************************************************************
The Gradle failure may have been because of AndroidX incompatibilities in this Flutter app.
See https://goo.gl/CP92wY for more information on the problem and how to fix it.
*******************************************************************************************
Finished with error: Gradle task assembleDebug failed with exit code 1"

Apologies if the fix is basic. This is my first day doing app dev xD

Cheers!

Flojolomo, 2019-11-16:
That's easy to fix. Add the line in file android/src/build.gradle: 
'implementation 'androidx.legacy:legacy-support-core-ui:1.0.0-alpha1''

It has to be subordinated to the object dependencies.
I would like to create a merge request, but have not found a solution to push my branch 😆

rubberbird, 2019-11-21:
> That's easy to fix. Add the line in file android/src/build.gradle:
> 'implementation 'androidx.legacy:legacy-support-core-ui:1.0.0-alpha1''
> 
> It has to be subordinated to the object dependencies.
> I would like to create a merge request, but have not found a solution to push my branch 😆

Hmm.. didn't work for me. Same issue.

enricobenedos, 2020-01-08:
Is your app using AndroidX ?

pauldemarco, 2020-04-02:
This should be resolved in the latest version.
Please make sure to update your project by deleting the Android and iOS folders and repairing with:
`flutter create .`

This is assuming you have no custom changes to these directories.

Please reopen if you are still having trouble, or stop into the discord for some help.

closed 2020-04-02 by pauldemarco