pauldemarco/flutter_blue #444: The Gradle failure may have been because of AndroidX incompatibilities in this Flutter app

yiguozhen0510, 2019-11-14:
when my project build,i got the peoblem "The Gradle failure may have been because of AndroidX incompatibilities in this Flutter app".angone has got it?help!!!

Megadec, 2019-11-14:
the latest version of this plugin now effectively only works with Flutter v1.10.15 which at this time of writing is the unstable dev channel.

#415 (comment)

ride4sun, 2020-02-13:
I still can not build the Example:

```
[√] Flutter (Channel stable, v1.12.13+hotfix.7, on Microsoft Windows [Version 10.0.19041.84], locale en-US)
    • Flutter version 1.12.13+hotfix.7 at c:\sdks\flutter
    • Framework revision 9f5ff2306b (3 weeks ago), 2020-01-26 22:38:26 -0800
    • Engine revision a67792536c
    • Dart version 2.7.0

 
[√] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
    • Android SDK at C:\Users\ride4\AppData\Local\Android\sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-28, build-tools 28.0.3
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b03)
    • All Android licenses accepted.

[√] Android Studio (version 3.5)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin version 43.0.1
    • Dart plugin version 191.8593
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b03)

[√] VS Code, 32-bit edition (version 1.41.1)
    • VS Code at C:\Program Files (x86)\Microsoft VS Code
    • Flutter extension version 3.7.1
```

**Here is the Error:**
```
Launching lib\main.dart on moto g 7 in debug mode...
[!] Your app isn't using AndroidX.
    To avoid potential build failures, you can quickly migrate your app by following the steps on https://goo.gl/CP92wY.
Running Gradle task 'assembleDebug'...
C:\Users\ride4\.gradle\caches\transforms-1\files-1.1\support-compat-28.0.0.aar\4f7ab4c6fa2922a87fc36a5d679eb841\res\values\values.xml:133:5-70: AAPT: error: resource android:attr/fontVariationSettings not found.

C:\Users\ride4\.gradle\caches\transforms-1\files-1.1\support-compat-28.0.0.aar\4f7ab4c6fa2922a87fc36a5d679eb841\res\values\values.xml:133:5-70: AAPT: error: resource android:attr/ttcIndex not found.

C:\Projects\flutter_blue\example\build\app\intermediates\incremental\mergeDebugResources\merged.dir\values\values.xml:84: error: resource android:attr/fontVariationSettings not found.
C:\Projects\flutter_blue\example\build\app\intermediates\incremental\mergeDebugResources\merged.dir\values\values.xml:84: error: resource android:attr/ttcIndex not found.
error: failed linking references.

Failed to execute aapt
com.android.ide.common.process.ProcessException: Failed to execute aapt
	at com.android.builder.core.AndroidBuilder.processResources(AndroidBuilder.java:796)
	at com.android.build.gradle.tasks.ProcessAndroidResources.invokeAaptForSplit(ProcessAndroidResources.java:551)
	at com.android.build.gradle.tasks.ProcessAndroidResources.doFullTaskAction(ProcessAndroidResources.java:285)
	at com.android.build.gradle.internal.tasks.IncrementalTask.taskAction(IncrementalTask.java:109)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:73)
	at org.gradle.api.internal.project.taskfactory.IncrementalTaskAction.doExecute(IncrementalTaskAction.java:50)
	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:39)
	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:26)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$1.run(ExecuteActionsTaskExecuter.java:131)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:300)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:292)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:174)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:90)
	at org.gradle.internal.operations.DelegatingBuildOperationExecutor.run(DelegatingBuildOperationExecutor.java:31)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeAction(ExecuteActionsTaskExecuter.java:120)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:99)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:77)
	at org.gradle.api.internal.tasks.execution.OutputDirectoryCreatingTaskExecuter.execute(OutputDirectoryCreatingTaskExecuter.java:51)
	at org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter.execute(SkipUpToDateTaskExecuter.java:59)
	at org.gradle.api.internal.tasks.execution.ResolveTaskOutputCachingStateExecuter.execute(ResolveTaskOutputCachingStateExecuter.java:54)
	at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:59)
[... 39 lines cut ...]
	at com.android.builder.internal.aapt.v2.QueueableAapt2.lambda$makeValidatedPackage$1(QueueableAapt2.java:179)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	... 1 more
Caused by: com.android.tools.aapt2.Aapt2Exception: AAPT2 error: check logs for details
[... cut ...]
Execution failed for task ':app:processDebugResources'.
> Failed to execute aapt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 3s
AndroidX incompatibilities may have caused this build to fail. Please migrate your app to AndroidX. See https://goo.gl/CP92wY.
Finished with error: Gradle task assembleDebug failed with exit code 1
```

pauldemarco, 2020-04-02:
This should be resolved with the latest published version 0.7.0.
Please reopen if you are still experiencing issues.

closed 2020-04-02 by pauldemarco

exeptionerror, 2021-06-24:
All Possible working solution added[ [Solved] Gradle failure may have been because of AndroidX incompatibilities in this Flutter app](https://fluttercorner.com/gradle-failure-may-have-been-because-of-androidx-incompatibilities-in-this-flutter-app/)