pauldemarco/flutter_blue #447: Write characteristic doesn't work after requestMtu

darkchaos86, 2019-11-14:
I faced an issue that after adding `await currentDevice.requestMtu(242)` to my code I cannot get write data to the characteristic. My code:
`        await currentDevice.requestMtu(242)
          characteristic.value.listen((value) {
            print("data received $value");
          });
          await writeCharacteristic.write(utf8.encode(command)).timeout(Duration(seconds: 3), onTimeout: () {
          print("onTimeout write");
        });`
Here is log output:
D/BluetoothGatt( 2556): configureMTU() - device: CA:6E:D6:5E:68:09 mtu: 242
I/flutter ( 2556): data received []
D/BluetoothGatt( 2556): onConfigureMTU() - Device=CA:6E:D6:5E:68:09 mtu=242 status=0
D/FlutterBluePlugin( 2556): [onMtuChanged] mtu: 242 status: 0
onTimeout write

But if I remove the line `await currentDevice.requestMtu(242)` I will be able to get the callback from the device.
Any thoughts?

darkchaos86, 2019-11-18:
The issue happened only after immediate execution of write request, but if I do it asynchronously, i.e. after some user action then everything works good

nmbus, 2020-02-20:
Had the same issue, actually hardly anything works right after the requestMtu call, in my case I was doing the services discovery after that call (even when using await to ensure the call completed). The call doesn't actually seem to return when the MTU is actually changed, that seems to be done later, so what I did to fix it was waiting on a Completer's future that is completed when the mtu changes (using a listener on the mtu property).

hungyip, 2020-03-30:
I have the same issue. After requestMtu call, characteristic read and write doesn't work after requestMtu. If I remove requestMtu, then it will work again.

efreibe, 2020-05-19:
Please view #579

savar, 2020-08-29:
having the same issue.. something is messing up but I cannot point into the right direction why this is the case..

savar, 2020-08-29:
> Had the same issue, actually hardly anything works right after the requestMtu call, in my case I was doing the services discovery after that call (even when using await to ensure the call [...]

@nmbus do you have a code snippet to share how you used a Completer for that?

xelons, 2020-09-01:
Same issue here, once requestMtu is called with or without await, then nothing works afterward, can't receive notifications either. I can't change MTU size, which makes all the communication protocol unusable.

efreibe, 2020-09-01:
I'm using this patch https://github.com/pauldemarco/flutter_blue/pull/579#issuecomment-628325875 to connect to bluetooth printers and setting mtu. It solved for me.
In android that code waits for the gatt event onMtuChanged before trigger completion, in that way you can write to bluetooth AFTER the mtu was set.

xelons, 2020-09-01:
> I'm using this patch [#579 (comment)](https://github.com/pauldemarco/flutter_blue/pull/579#issuecomment-628325875) to connect to bluetooth printers and setting mtu. It solved for me.
> In android that code waits for the gatt event onMtuChanged before trigger completion, in that way you can write to bluetooth AFTER the mtu was set.

Thanks, I already tried your patch, but unfortunately still experiencing the same issue. Once I call requestMtu, I can no longer write to the characteristic.
When I requestMtu using the nrf scanner app, it works perfectly, so I know its not my ble device at fault here.

efreibe, 2020-09-01:
I don't know what value you are passing in, but maybe it's a negotiation
problem.
What about passing lower values, let's say 20, then 40, then 100 and see if
anything changes. Search in the log panel for "[onMtuChanged]"

xelons, 2020-09-01:
> I don't know what value you are passing in, but maybe it's a negotiation problem. What about passing lower values, let's say 20, then 40, then 100 and see if anything changes. Search in the log panel for "[onMtuChanged]"

Im using  await widget.device.requestMtu(185); as soon as I call it, Characteristic.write does not work anymore.
I checked the Mtu and it is indeed changed to 185, so it seems like the request to change Mtu is successful, but write calls no longer works.
When I set it using nrf scanner app it works perfectly.

Bobby9292, 2020-09-02:
> Im using await widget.device.requestMtu(185); as soon as I call it, Characteristic.write stops working, no matter what I do.
> I checked the Mtu and it is indeed changed to 185, so it seems like the request to change Mtu is successful, but write calls no longer works.
> When I set it using nrf scanner app it works perfectly.

From what I understand of your explanation you are requesting a new MTU in the middle of your write operations ? 

If it's the case it's probably normal that your write will stop since you are asking device for a new MTU and awaiting for it to answer which should stop any other writing operations. 

In any case it's better to negotiate the MTU and wait for completion,  before starting any writing operations. 

Also if you already put the negotiation before writing, make sure that you copied the snippet @efreibe submitted in the correct file : 
- flutter_blue-0.7.2/lib/src/bluetooth_device.dart. 

Maybe sharing the bit of code and the error prompted in your terminal would allow us to help you more too.. 

GL

savar, 2020-09-02:
> Thanks, I already tried your patch, but unfortunately still experiencing the same issue. Once I call requestMtu, I can no longer write to the characteristic.
> When I requestMtu using the nrf scanner [...]

@xelons please be aware, the MTU is not necessarily the DATA MTU but maybe the ATT MTU (depends on iOS/Android and all other kind of things).. so after requesting an MTU of let's say 100, try to send only 97 bytes per char write as 3 bytes might still be required for the ATT header which is not consistently reported back. (the minimum ATT MTU is 23, but flutter_blue hardcodes initially an MTU of 20, thats why you don't have that issue with the initial MTU).

xelons, 2020-09-02:
@Bobby9292 I request the Mtu after I connect and before I do any discovery or writes. It seems like the write buffer is stuck as soon as requestMtu is called.
@savar Thanks, yes I know, this issue is only happening on Android, iOS works fine. My write command is just about 10 bytes, my received notifications are the ones needing 185 (around 180 of data or less).

One thing that is worth noting is that my characteristic properties are set to notify and write, I'm going to enable read property on it and see if that is going to solve the blocking issue, although I don't think it makes much sense.

xelons, 2020-09-02:
I found the solution to my issue, I had to enable Read property on my characteristic for the requestMtu to work and not block the write buffer. Hope this can be useful to someone.

savar, 2020-09-02:
> @Bobby9292 I request the Mtu after I connect and before I do any discovery or writes. It seems like the write buffer is stuck as soon as requestMtu is called.
> @savar Thanks, yes I know, this issue [...]

As iOS is ignoring your requestMtu (see the docs) there is no surprise that this works fine on iOS :wink: .. but good to hear that you found the solution!

savar, 2020-09-02:
also see https://github.com/pauldemarco/flutter_blue/pull/579#issuecomment-685448833

Atokulus, 2021-04-23:
Same problem in my case.