pauldemarco/flutter_blue #453: Scan doesn't always show existing, connectable devices

eliseyOzerov, 2019-11-26:
This is my code for getting the list of valid devices. 

`flutterBlue.scanResults.listen((scanResultList){        
        setAvailableDevices(scanResultList);
      }, onDone: (){
        print("done listening to scanresults");
      }, onError: (err){
        print("Error listening to scanResults, $err");
      });
      print("Currently not scanning, starting scan");
      _scanSubscription = flutterBlue.scan(timeout: Duration(seconds: 1)).listen((scanResult) {
        print("${scanResult.device.name}");
        if(scanResult.device.id.toString() == prefs.getString(Constants.RIDERSENSORID)) {
          if(isAutomatic) connectDevice(scanResult.device, isAutomatic, ConnectionType.rider, onError);  
          if(favoriteRiderDeviceId == null || favoriteRiderDeviceId == "") setFavoriteRiderDeviceId(scanResult.device.id.toString());        
        }        
        else if(scanResult.device.id.toString() == prefs.getString(Constants.HORSESENSORID)) {
          if(isAutomatic) connectDevice(scanResult.device, isAutomatic, ConnectionType.horse, onError);
          if(favoriteHorseDeviceId == null || favoriteHorseDeviceId == "") setFavoriteHorseDeviceId(scanResult.device.id.toString());
        }
      }, onDone: ()async {
        print("done scanning");
        await cancelScan();
      }, onError: (msg){
        print("Error: ${msg.toString()}");
      });
      _scanSubscription.onError((err){
        print("Error has occured while scanning");
      });`

`@action
  void setAvailableDevices(List<ScanResult> list){
    availableDevices = list.where((scanResult) => isHeartRateSensor(scanResult.device)).map((scanResult) => scanResult.device).toList();
    if(!availableDevices.any((device)=>device == riderConnectedDevice && riderConnectedDevice != null)) availableDevices.add(riderConnectedDevice);
    if(!availableDevices.any((device)=>device == horseConnectedDevice && horseConnectedDevice != null)) availableDevices.add(horseConnectedDevice);
    print(list.length);
  }

  /// Check if the given `BluetoothDevice` is a listed heart rate sensor.
  bool isHeartRateSensor(BluetoothDevice device){
    return validSensorNames.any((value)=>device.name.contains(value));
  }`

Here is the "modified" scan function inside the plugin. Basically what happens is i the last print statement I get is "before yield
`Stream<ScanResult> scan({
    ScanMode scanMode = ScanMode.lowLatency,
    List<Guid> withServices = const [],
    List<Guid> withDevices = const [],
    Duration timeout,
  }) async* {
    print("starting async scan process");
    var settings = protos.ScanSettings.create()
      ..androidScanMode = scanMode.value
      ..serviceUuids.addAll(withServices.map((g) => g.toString()).toList());
    print("start of scan");
    if (_isScanning.value == true) {
      print("another scan is running");
      throw Exception('Another scan is already in progress.');
    }
    print("before isScanning set to true");
    // Emit to isScanning
    _isScanning.add(true);

    final killStreams = <Stream>[];

    killStreams.add(_stopScanPill);
    if (timeout != null) {
      print("timeout isnt null, adding killstream");
      killStreams.add(Observable.timer(null, timeout));
    }

    print("before scanresults are cleared");
    // Clear scan results list
    _scanResults.add(<ScanResult>[]);

    try {
      print("before invoking channel method startScan");
      var result = await _channel.invokeMethod('startScan', settings.writeToBuffer());
      print("done with startScan, result: $result");
    } catch (e) {
      print('Error starting scan.');
      _stopScanPill.add(null);
      _isScanning.add(false);
      throw e;
    }
    print("before yield");
    try {yield* Observable(FlutterBlue.instance._methodStream.where((m) => m.method == "ScanResult").map((m) => m.arguments))
        .takeUntil(Observable.merge(killStreams))
        .doOnDone((){stopScan; print("stopped scan");})
        .map((buffer) => new protos.ScanResult.fromBuffer(buffer))
        .map((p) {
          print("beginning of yield");
      final result = new ScanResult.fromProto(p);
      final list = _scanResults.value;
      int index = list.indexOf(result);
      if (index != -1) {
        list[index] = result;
      } else {
        list.add(result);
      }
      print("Scan results length: ${_scanResults.length}");
      _scanResults.add(list);
      return result;
    });} catch (e){
      print("error $e");
      throw e;
    }
  }`

I've also added a print statements all over the scan method inside the plugin and a try catch around yield* Observer method.

The problem I'm having is that with no obvious reasoning, scan doesn't always show devices that are obviously on and in range. It also doesn't always show devices that are already connected to the phone. I've spent about a day trying to figure it out, but with no success. 

IsScanning boolean is always false in this case, so that isn't the issue. There is 0 error outputs, and no caught exceptions. The scanResults are just empty and scanning stops. Here is sample output from print statements when the scanning stops with no results:

`I/flutter (27076): Scan was called in BluetoothStore
I/flutter (27076): setting scanResults listener
I/flutter (27076): Currently not scanning, starting scan
I/flutter (27076): false
I/flutter (27076): 14
I/flutter (27076): starting async scan process
I/flutter (27076): start of scan
I/flutter (27076): before isScanning set to true
I/flutter (27076): timeout isnt null, adding killstream
I/flutter (27076): before scanresults are cleared
I/flutter (27076): before invoking channel method startScan
I/flutter (27076): 0
I/flutter (27076): 0
I/flutter (27076): done with startScan, result: success
I/flutter (27076): before yield
I/flutter (27076): stopped scan
I/flutter (27076): done scanning
I/flutter (27076): finished cancelling scan`

Any clues as to what might be happening? Thanks

alejo9719, 2019-12-04:
Did you solve it?

ifredom, 2020-01-13:
Did you solve it?

eliseyOzerov, 2020-01-13:
I shows the devices now, but I don't know what caused the bug, so I don't really have a fix

pauldemarco, 2020-04-02:
Will close this for now. Please reopen if there are any more developments.

closed 2020-04-02 by pauldemarco