pauldemarco/flutter_blue #454: 0.6.3+1: iOS: Can't import package

mikaelj, 2019-11-28:
I get jiggly lines under

    import "package:flutter_blue/flutter_blue.dart"

I depend on a bunch of packages, but they all work. This one does not.

Here's what the doctor says.

    [✓] Flutter (Channel beta, v1.11.0, on Mac OS X 10.14.5 18F132, locale en-SE)
        • Flutter version 1.11.0 at /Users/mikaelj/Library/flutter
        • Framework revision 856a90e67c (3 weeks ago), 2019-11-08 18:00:01 -0800
        • Engine revision af04338413
        • Dart version 2.7.0

    [✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
        • Android SDK at /opt/android-sdk
        • Android NDK location not configured (optional; useful for native profiling support)
        • Platform android-28, build-tools 28.0.3
        • ANDROID_HOME = /opt/android-sdk
        • ANDROID_SDK_ROOT = /opt/android-sdk
        • Java binary at: /Applications/Android Studio 3.5 Preview.app/Contents/jre/jdk/Contents/Home/bin/java
        • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-1343-b01)
        • All Android licenses accepted.

    [✓] Xcode - develop for iOS and macOS (Xcode 11.0)
        • Xcode at /Applications/Xcode.app/Contents/Developer
        • Xcode 11.0, Build version 11A420a
        • CocoaPods version 1.8.4

    [✓] Android Studio
        • Android Studio at /Applications/Android Studio 3.5 Preview.app/Contents
        • Flutter plugin version 36.1.1
        • Dart plugin version 183.6270
        • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-1343-b01)

    [✓] VS Code (version 1.40.1)
        • VS Code at /Applications/Visual Studio Code.app/Contents
        • Flutter extension version 3.6.0

    [✓] Connected device (1 available)
        • iPhone SE • ... • ios • com.apple.CoreSimulator.SimRuntime.iOS-12-2 (simulator)

    • No issues found!

Nailer, 2019-11-29:
I'm also still getting:

```
'flutter_blue/FlutterBluePlugin.h' file not found
    #import <flutter_blue/FlutterBluePlugin.h>
```
when trying to build to iOS with the newest version of everything. See the unresolved issue in #386

mikaelj, 2019-11-29:
To be clear: this post adresses the stage before anything is even built.
It's as if "flutter pub get" wasn't run or the package wasn't in pubspec.yaml

mateuszjablonski, 2019-11-30:
@mikaelj try opening `Runner.xcworkspace`, searching for `FlutterBluePlugin.h` file in _Project Navigator_, and then setting it's _Target Membership_ for `flutter_blue` to _Public_. And please let me know then, if the issue still occurs.

mikaelj, 2019-11-30:
@mateuszjablonski 
1. That file isn't visible from Xcode -- the path is ./Pods/Headers/Public/flutter_blue/FlutterBluePlugin.h but Xcode hides the directory Pods/Headers

2. My problem happens on the Dart end, before the iOS backend is ever being touched. (It really doesn't matter if it's iOS or Android, so I'll change the header now).

See screenshot.

[image]

mikaelj, 2019-11-30:
...and completely magically, Dart could find the header file. I tried running it on simulator.

Closing as I have absolutely no idea what the cause was and why it works now :-/

closed 2019-11-30 by mikaelj