pauldemarco/flutter_blue #455: Unable to write characteristic

PrzemyslawSarnacki, 2019-11-28:
Writing a characteristic with a write method is impossible even while trying on example. The same task is possible while trying it with nRF Connect app. On every try this exception occurs:

> E/flutter (15693): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null)
E/flutter (15693): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
E/flutter (15693): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:321:33)
E/flutter (15693): <asynchronous suspension>
E/flutter (15693): #2      BluetoothCharacteristic.write (package:flutter_blue/src/bluetooth_characteristic.dart:123:10)
E/flutter (15693): #3      DeviceScreen._buildServiceTiles.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:flutter_joystick/main.dart:179:37)
E/flutter (15693): #4      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:706:14)
E/flutter (15693): #5      _InkResponseState.build.<anonymous closure> (package:flutter/src/material/ink_well.dart:789:36)
E/flutter (15693): #6      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:182:24)
E/flutter (15693): #7      TapGestureRecognizer.handleTapUp (package:flutter/src/gestures/tap.dart:486:11)
E/flutter (15693): #8      BaseTapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:264:5)
E/flutter (15693): #9      BaseTapGestureRecognizer.acceptGesture (package:flutter/src/gestures/tap.dart:236:7)
E/flutter (15693): #10     GestureArenaManager.sweep (package:flutter/src/gestures/arena.dart:156:27)
E/flutter (15693): #11     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:222:20)
E/flutter (15693): #12     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:198:22)
E/flutter (15693): #13     GestureBinding._handlePointerEvent (package:flutter/src/gestures/binding.dart:156:7)
E/flutter (15693): #14     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:102:7)
E/flutter (15693): #15     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:86:7)
E/flutter (15693): #16     _rootRunUnary (dart:async/zone.dart:1138:13)
E/flutter (15693): #17     _CustomZone.runUnary (dart:async/zone.dart:1031:19)
E/flutter (15693): #18     _CustomZone.runUnaryGuarded (dart:async/zone.dart:933:7)
E/flutter (15693): #19     _invoke1 (dart:ui/hooks.dart:273:10)
E/flutter (15693): #20     _dispatchPointerDataPacket (dart:ui/hooks.dart:182:5)

pauldemarco, 2020-04-02:
Can you supply me with the service, characteristic, and descriptors?

PrzemyslawSarnacki, 2020-04-02:
I've done that in following manner:

`    void writeChar() async {
      List<BluetoothService> services = await device.discoverServices();
      services.forEach((service) {
        service.characteristics.forEach((character) {
          if (character.uuid.toString() == WRITECHARACTERISTIC_UUID) {
            character.write([0x01]);
          }
        });
      });
    }`

And variable which contains characteristic: 
`  static const String WRITECHARACTERISTIC_UUID =
      "013dc1df-9b8c-4b5c-949b-262543eba78a";`

pauldemarco, 2020-04-02:
What properties does the characteristic have enabled?  Does it have any descriptors?

windswrold, 2020-08-26:
我也遇到了此问题，我是快速写入时会遇到。

windswrold, 2020-08-26:
hi ,I also encountered this problem, which I encountered when writing quickly.
E/flutter (26820): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null)
E/flutter (26820): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
E/flutter (26820): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:156:18)
E/flutter (26820): <asynchronous suspension>
E/flutter (26820): #2      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:329:12)
E/flutter (26820): #3      BluetoothCharacteristic.write (package:flutter_blue/src/bluetooth_characteristic.dart:121:10)
E/flutter (26820): #4      BlueHelp._send (package:HolderWallet/blue/BlueHelp.dart:383:28)

PrzemyslawSarnacki, 2021-02-07:
I bought device from other manufacturer (ESP32) and tested the same characteristic and code base and it worked like a charm. So I suppose that this is not a problem of a flutter blue plugin. And also I can say that nRF Connect was writing characteristic in server mode which is totally different.  I'm sorry for creating this issue.

windswrold, 2021-02-08:
> I bought device from other manufacturer (ESP32) and tested the same characteristic and code base and it worked like a charm. So I suppose that this is not a problem of a flutter blue plugin. And [...]

You can close issue

closed 2021-02-08 by PrzemyslawSarnacki