pauldemarco/flutter_blue #456: Frame Lost

Nicholas-Luoyi, 2019-11-29:
I've got some frame lost in receiving data with flutter_blue 0.6.2.

The master bluetooth device will send me a upcounting frame every 10ms. And I found that the flutter_blue won't handle data to me on receiving the data. But it will keep all the data for up to about 50ms. Let's see this.

In FlutterBluePlugin.java I can see it will log when characteristic changed.
`log(LogLevel.DEBUG, "[onCharacteristicChanged] uuid: " + characteristic.getUuid().toString());`

And in my flutter code. I will also log something on receiving the data.
`bluetoothCharacteristic.value.listen((bytes){debugPrint('Receive Frame: $bytes');});`

So that means if the master bluetooth device send something. We will see a "onCharacteristicChanged" and a "Receive Frame" on terminal one by one.

But the fact is that we will have some "onCharacteristicChanged" at once, and after that we will have the same numbers of "Receive Frame" as "onCharacteristicChanged". Following the pictures below:

[image]

And I found some frames were missed between the two rectangles. It seems that flutter_blue will miss something when it is handling data to me.

JesusAlexlara, 2019-12-11:
I have the same problem of losing frames but on iOS, did you find a solution?

Nicholas-Luoyi, 2019-12-25:
No solution at present. Just have to delay the interval between the frames.

Kanonym, 2020-03-24:
Hey, I have the same problem. I am sending some data to my phone, at the moment with a speed of 50 Hz but as soon as i get over this threshol , I am losing frames in my app. i receive the data but the rest of the code doesn' execute anymore.

Nicholas-Luoyi, 2020-03-24:
> Hey, I have the same problem. I am sending some data to my phone, at the moment with a speed of 50 Hz but as soon as i get over this threshol , I am losing frames in my app. i receive the data but the rest of the code doesn' execute anymore.

How many bytes do you have in one frame

Nicholas-Luoyi, 2020-03-25:
> Hey, I have the same problem. I am sending some data to my phone, at the moment with a speed of 50 Hz but as soon as i get over this threshol , I am losing frames in my app. i receive the data but the rest of the code doesn' execute anymore.

I think you should check your MTU. If your frame length is larger than the MTU. You should use BluetoothCharacteristic.write(List<int> bytes, withoutResponse: false). Or you can only get a frame with length of MTU size.