pauldemarco/flutter_blue #458: set_notification_error 0.6.2

chaochaox1990, 2019-12-02:
E/flutter ( 9139): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(set_notification_error, could not locate CCCD descriptor for characteristic: 00002a00-0000-1000-8000-00805f9b34fb, null)
E/flutter ( 9139): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
E/flutter ( 9139): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:316:33)
E/flutter ( 9139): <asynchronous suspension>
E/flutter ( 9139): #2      BluetoothCharacteristic.setNotifyValue (package:flutter_blue/src/bluetooth_characteristic.dart:157:10)
E/flutter ( 9139): <asynchronous suspension>
E/flutter ( 9139): #3      DeviceScreen._buildServiceTiles.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:flutter_blue_example/main.dart:170:27)
E/flutter ( 9139): #4      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:654:14)
E/flutter ( 9139): #5      _InkResponseState.build.<anonymous closure> (package:flutter/src/material/ink_well.dart:729:32)
E/flutter ( 9139): #6      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:182:24)
E/flutter ( 9139): #7      TapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:365:11)
E/flutter ( 9139): #8      TapGestureRecognizer.acceptGesture (package:flutter/src/gestures/tap.dart:312:7)
E/flutter ( 9139): #9      GestureArenaManager.sweep (package:flutter/src/gestures/arena.dart:156:27)
E/flutter ( 9139): #10     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:222:20)
E/flutter ( 9139): #11     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:198:22)
E/flutter ( 9139): #12     GestureBinding._handlePointerEvent (package:flutter/src/gestures/binding.dart:156:7)
E/flutter ( 9139): #13     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:102:7)
E/flutter ( 9139): #14     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:86:7)
E/flutter ( 9139): #15     _rootRunUnary (dart:async/zone.dart:1136:13)
E/flutter ( 9139): #16     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter ( 9139): #17     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter ( 9139): #18     _invoke1 (dart:ui/hooks.dart:263:10)
E/flutter ( 9139): #19     _dispatchPointerDataPacket (dart:ui/hooks.dart:172:5)
E/flutter ( 9139): 

pls help me check,  thanks

closed 2019-12-05 by chaochaox1990

cza100, 2019-12-06:
@chaochaox1990 same error, have you fixed it?

chaochaox1990, 2019-12-27:
> @chaochaox1990 same error, have you fixed it?
yes I already fixed. All I do just change the value of CCCD.