pauldemarco/flutter_blue #461: Scan example in official README throws error

alejo9719, 2019-12-04:
The example in the official README, also shown in pub.dev throws an error (The getter 'device' isn't defined for the class 'List<ScanResult>'.

I think the example reads the device as if the scan were initiated using the scan method instead of the startScan method.

I tried this instead:

```
// Start scanning
flutterBlue.startScan(timeout: Duration(seconds: 4));

// Listen to scan results
var subscription = ftBlue.scanResults.listen((scanResultsList) {
      // Last element in scanResultsList is the result of the last scan
      var device = scanResultsList.last.device; // Get a device from the stream
      foundDevices.add(device);
      print('${device.name} found! rssi: ${scanResultsList.last.rssi}'); //DEBUG
    });

// Stop scanning
flutterBlue.stopScan();
```

It solves that issue, but the list of devices is empty, so "device" does not exist. Furthermore, doing this:
```
// Start scanning and wait for its completion
var scanResultsList = await ftBlue.startScan(timeout: Duration(seconds: waitSeconds));
print(scanResultsList);
```
prints an empty list. I've tried with waitSeconds  equal to 4, 15 and 30.

agnoam, 2020-02-04:
I have the same issue... 
Why search function returns an empty list ?

alejo9719, 2020-02-04:
> I have the same issue...
> Why search function returns an empty list ?

You have to enable the location service in your mobile device for the bluetooth low energy devices to be detected. Also, take into account that the scan function is executed for the specified duration and results are returned via a stream at the end of search.

pauldemarco, 2020-04-02:
Fixed in recent commit.

closed 2020-04-02 by pauldemarco