pauldemarco/flutter_blue #462: BluetoothDevice:state bad behaviour when reconnecting

boskokg, 2019-12-10:
I think we have a problem with the device reconnection with the next code:

  /// The current connection state of the device
  Stream<BluetoothDeviceState> get state async* {
    yield await FlutterBlue.instance._channel
        .invokeMethod('deviceState', id.toString())
        .then((buffer) => new protos.DeviceStateResponse.fromBuffer(buffer))
        .then((p) => BluetoothDeviceState.values[p.state.value]);

    yield* FlutterBlue.instance._methodStream
        .where((m) => m.method == "DeviceState")
        .map((m) => m.arguments)
        .map((buffer) => new protos.DeviceStateResponse.fromBuffer(buffer))
        .where((p) => p.remoteId == id.toString())
        .map((p) => BluetoothDeviceState.values[p.state.value]);
  }

The part that makes the problem is "where((p) => p.remoteId == id.toString())".  This part yields the states of the previous connection of the same device.

Any advice how I can listen only to the states of the new connection?

MrHazee, 2020-01-13:
Hi!
I have the same problem. Did you find a soloution? 
/Rasmus

pauldemarco, 2020-04-02:
@boskokg  can you describe the symptoms you are facing? What is the problem you are seeing?

boskokg, 2020-04-03:
@pauldemarco after the reconnection to the same device I just call device.connect. problem is that streams are not cleared on reconnection so I receive the values from the old connection.

To prevent this I had to create a map of streams in order to skip old values from the old connection:

  Map<Guid, StreamSubscription> _servicesSubscription = {};
  Map<BluetoothDevice, StreamSubscription> _stateListeners = {};

and in the reconnect method I have this part of the code (I am pasting the android part where I don't use scan since I have the mac address of the device):

      protos.BluetoothDevice dev = protos.BluetoothDevice()..remoteId = address;
      _gattClient = BluetoothDevice.fromProto(dev);

      if (!_stateListeners.containsKey(_gattClient)) {
        _stateListeners[_gattClient] = _gattClient.state
            .skip(1)
            .listen((BluetoothDeviceState newState) => _onConnectionStateChange(_gattClient, newState));
      }
      _gattClient.connect();

pauldemarco, 2020-04-03:
Are you seeing duplicate notifications for instance? Does this N+1 for each reconnect?

boskokg, 2020-04-03:
You can see that I was forced to do one stream listening per device during the app lifetime.

fyredave, 2020-06-03:
I'm also having this issue.

On a Google Pixel 2 XL running Android 10

When I disconnect I call 
`await _stateSubscription?.cancel();
_stateSubscription = null;`

When I reconnect a few seconds later I call
`_stateSubscription = _device.state.listen(_handleBLEStateChanged);`

after that _handleBLEStateChanged is fired twice. If I disconnect and reconnect it fires a third time. 

It continues to fire an additional duplicate event every time I reconnect

I have tried using single stream subscription with  onPause and onResume but it has the same effect. It's possible it's the same error in the stream logic that was discovered here:
https://github.com/pauldemarco/flutter_blue/issues/525