pauldemarco/flutter_blue #463: blutter blue for macos

ekuleshov, 2019-12-11:
With Flutter 1.12 adding macos target as beta, it would be great to have macos support in flutter blue.

xvrh, 2019-12-18:
I tried this quickly and this is actually easy to add. 
The whole iOS code can be reused to run on macOS

Instructions:
- Copy/paste the `ios` folder in a `macos` folder
- In `FlutterBluePlugin`, change the import `<Flutter/Flutter.h>` to `<FlutterMacOS/FlutterMacOS.h>`
- Change the flutter section in the pubspec.yaml  to:
```yaml
flutter:
  plugin:
    platforms:
      android:
        package: com.pauldemarco.flutterblue
        pluginClass: FlutterBluePlugin
      ios:
        pluginClass: FlutterBluePlugin
      macos:
        pluginClass: FlutterBluePlugin
```
- In the target application project, open the macos folder in XCode and in Runner > Signing & Capabilities > App Sandbox > check Bluetooth.

Ahmadre, 2020-07-11:
can we please have this as a PR?

zwells, 2020-07-23:
Since macOS will be able to run iOS apps coming this Fall I'm not sure if it's actually needed anymore...

CCBDS, 2021-11-24:
So if you have this problem, first of all, turn on the swtich under Developer Mode on windows 10 system version