pauldemarco/flutter_blue #464: writes on a characteristic also marked for notifications are failing for some android devices

khainke, 2019-12-11:
There appears to be a problem, when performing (fast and multiple) writes on a characteristic also marked as notifying characteristic. (write with response, awaited in a queue, set notify value for the same characteristic on true) 

To be specific, for every characteristic my implementation does only ever do 1 concurrent write on a characteristic, with response, awaits that call and only starts the next write when the previous has finished (writes are queued). But there are different queues for every characteristic/device. At the same time im communicating with multiple (2,3) bluetooth devices at once, so that they might overlap. This works fine for every devices, when the characteristics written on are not also marked for notifications. But when the characteristic written on, is also marked for notifications, it works fine for every ios device, and for some android devices, but for example, the problem occurs on a Pixel3a in my tests. For this phone, at some point, a write and every following write will fail, without a specific error. The onCharacteristicWrite debug log message shows, that the bluetooth gatt state was 2 meaning GATT_READ_NOT_PERMITTED. 

I was googling around, and it seems, that Androids implementation for Bluetooth are somewhat fragile and that this is a common problem, ios does queue the writes, android does not, maybe some android variants (samsung, etc.) do, which would explain, why it works on some android devices, but fails on others. Some links:
https://medium.com/@martijn.van.welie/making-android-ble-work-part-3-117d3a8aee23
https://punchthrough.com/android-ble-development-tips/
https://stackoverflow.com/questions/32389351/how-to-write-characteristic-fast-and-stable-for-ble-in-android/32402515

pauldemarco, 2020-04-02:
@khainke  are you feeling this may be a quirk with the Android Bluetooth libraries?  Are you seeing this on only certain devices, while other devices work great?

khainke, 2020-04-02:
Yes, it really depends on the individual android device, certain models (SGA40, SGS7) dont have any problems, while other (Pixel3a, Mi9T) need to be implemented with a delay and a mutex to make it work or else they might fail 80% of BluetoothCharacteristic.write() calls, when sending in fast succession, while also listening for notifications. 
Also sometimes you see the "faulty" devices receiving answers from the connected device on the characteristic, while the write call is still not concluded/returned. But on the "normal" devices, the behavior is as expected, where the write call returns and then the answer is received.

It could very well be some deeply hidden android quirk, or how different distribution of android / phone models handle the callbacks with notifying and writing, and that some may be more leniant towards consistent, chronical order of request and callbacks.