pauldemarco/flutter_blue #465: Characteristic Read ERROR.

borisllona, 2019-12-22:
I want simply to read the values of the characteristics of my services. But I'm getting this error:
[image]

My Code `discoverServices() async {
    List<BluetoothService> services = await device.discoverServices();
    services.forEach((service) async {
      // do something with service
      if (service.uuid.toString() == SERVICE_UUID) {
        var characteristics = await service.characteristics;
        for(BluetoothCharacteristic c in characteristics) {
          //if (characteristic.uuid.toString() == CHARACTERISTIC_UUID) {
            targetCharacteristic = c;
            List<int> value = await c.read();
            print(value);
          //}
        };
      }
    });
  }`
[image]

Any idea of what's happening here? My head's gonna explote.
Thanks!!

closed 2019-12-29 by borisllona

aawhitfield, 2020-01-04:
Did you find out what the issue was? I'm having the same problem.

borisllona, 2020-01-04:
No... I'm still searching for more info

aawhitfield, 2020-01-04:
Thanks. I can't get it to work either but I set the notify value and listened for changes and got the data from there without calling read().

`await characteristic.setNotifyValue(true);
characteristic.value.listen((value) {
    // do something with new value
});`

borisllona, 2020-01-04:
Thanks mate! We keep in touch if we know any news about it. About the code
you sent me, it works to read a value?

aawhitfield, 2020-01-04:
Yeah. I always got an error trying to call read on the characteristic. I'm trying to get a blood pressure reading off a bluetooth blood pressure cuff. I used the code I sent you and in the .listen() function I set it to print(value) and you could see the values change while the machine was taking blood pressure and then when it was done, the variable 'value' had the final blood pressure reading as one of the integers in the List<int> that it returned and I could use the data that way and didn't need to call characteristic.read() in the end,