pauldemarco/flutter_blue #467: How to save a device instance for hot reload?

borisllona, 2019-12-29:
I'm working on an APP that uses a page that gets a device passed by parameter. The first time i run it it works perfectly because its redirected from another function that also passes the device as a parameter. But when I hot reload or I just minimize the app I get: **device is called on null**. I understand that its because it doesnt receive nothing by parameter when its hot reloaded. Any suggestion what I have to do in that case?
Thanks community!

bobosette, 2020-01-14:
I have a similar problem. I had a pageview but i can't pass the device by parameter. Is there a way to save the device instance?

borisllona, 2020-01-14:
If you find the solution tell me, I can't move along

bobosette, 2020-01-14:
ok

KevinRohn, 2020-01-22:
Same problem here :) 
Has someone a workaround to fix this? 
It costs me a lot of time during the development process. So let me know if someone has a solution for it.

pauldemarco, 2020-04-02:
Interesting, I feel this isn’t a flutter_blue issue but more a state management question in flutter.

If you disagree, please explain why and re-open the issue.

Thanks.

closed 2020-04-02 by pauldemarco