pauldemarco/flutter_blue #470: There is a 36 byte data to send out, so I want to set the requestmtu, but I can't connect to the device after setting.

cander0815, 2020-01-04:
code: 
`
     await device.connect(autoConnect: false);
      statcListen?.cancel();
        statcListen = device.state.listen((data) async {
        switch (data) {
          case BluetoothDeviceState.connected:
            isConnect = true;
            print('success');
            await device.requestMtu(64);
            print('Modify character length');
            connectIndex = 0;
            if (writeCharacteristics == null || readCharacteristics == null) {
              findService(() async {
                cb(true);
              });
            }
            break;
          case BluetoothDeviceState.disconnected:
            isConnect = false;
            print('disconnect');
            break;
          default:
            break;
        }
      });
`
logs:
`/BluetoothGatt(20062): onConfigureMTU() - Device=20:19:11:07:14:15 mtu=23 status=0
D/FlutterBluePlugin(20062): [onMtuChanged] mtu: 23 status: 0
D/BluetoothGatt(20062): onConnectionUpdated() - Device=20:19:11:07:14:15 interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt(20062): onConnectionUpdated() - Device=20:19:11:07:14:15 interval=24 latency=0 timeout=400 status=0`
What should I do to send it completely???

cander0815, 2020-01-04:
It is not necessary to set MTU for 36 byte data, and the await is removed when sending;
before: 
```
await writeCharacteristics.write(reqData);
```

now: 
```
writeCharacteristics.write(reqData);
```

closed 2020-01-04 by cander0815

mericgerceker, 2021-03-29:
Hi! I'm having the same problem. How did you find a solution?