pauldemarco/flutter_blue #471: Incorrect dispose of flutter blue, maybe missing dispose method ?

diegomgarcia, 2020-01-04:
Hi I've been playing with flutter_blue and instantiate it at onInitState to auto connect to an specific device, its working everything fine until the user hits the back button, after that I start to get the follow message logged in the console: 

D/FlutterNativeView( 2516): FlutterView.send called on a detached view, channel=plugins.pauldemarco.com/flutter_blue/methods

First I noted that my disconnect method wasn't being called for the dispose, then I found this huge thread about it https://github.com/flutter/flutter/issues/28134

Some recommendations is to implement it a method at the plugin to handle it nicely when the view is disposed, others recommend to handle it in the view implementing the WidgetsBindingObserver and intercept the state changes. 

What do you guys think about it ?

Regards,
Diego Garcia

alexsullivan114, 2021-12-07:
@diegomgarcia Did you get any further traction with this issue? I'm running into what I think is a similar problem and I'm not sure how to deal with it.

diegomgarcia, 2021-12-07:
Hi @alexsullivan114 to overcome this problem I've add the follow method: 

```
void disconnectBLEDevice() {
    print("Disconnecting");

    deviceStateSubscription.cancel();

    if (bleDevice != null)
      bleDevice.disconnect();
  }
```

on initstate I check if it exists and disconnect if needed. using this code: 

```
@override
  void initState() {
    super.initState();

    LocalPrefs.hasDeviceSaved().then((r) {
      setState(() {
        _hasDeviceSaved = r;
      });

      if (_hasDeviceSaved) {

        if (bleDevice != null)
          disconnectBLEDevice();

        connectBLEDevice();
      }
    });
  }

```

to finish just disconnect on dispose as the follow: 

```
@override
  void dispose() {
    print("disposing");

    disconnectBLEDevice();

    super.dispose();
  }

```

Let me know if this help you.

alexsullivan114, 2021-12-07:
@diegomgarcia So you're basically saving off whether you previously connected to the device in shared preferences, then disconnecting on init if you discover that you had connected? Makes sense!