pauldemarco/flutter_blue #473: Want to connect to a specific device

bobosette, 2020-01-06:
Hi everybody. I'm implementing flutter blue on my project. In my app I just would like to connect to a specific device and, like in this comment https://github.com/pauldemarco/flutter_blue/issues/125#issuecomment-432684180 , i could put a control on the device name.
Unfortunately, in the main.dart i don't find the method _startScan().
Somebody has any idea???

Thank you

jeremy110, 2020-01-13:
I know how to show the specific devices.You can find the codes on flutter blue widget.
[image]
[image]

bobosette, 2020-01-13:
Thank u. I already found a solution

closed 2020-01-13 by bobosette