pauldemarco/flutter_blue #477: RxDart Version incompability

saulopef, 2020-01-10:
Recently we need to upgrade the version of rxdart library, but during this process this massage have ocurred:

> [fitwiz_app] flutter packages get
> Running "flutter pub get" in fitwiz_app...                      
> Because flutter_blue >=0.6.0 depends on rxdart ^0.22.0 and fitwiz_app depends on rxdart 0.23.1, flutter_blue >=0.6.0 is forbidden.
> 
> So, because fitwiz_app depends on flutter_blue ^0.6.3+1, version solving failed.
> pub get failed (1; So, because fitwiz_app depends on flutter_blue ^0.6.3+1, version solving failed.)
> exit code 1

Aparently, the flutter_blue depends the rxdart 0.22.0 version, but when we remove the rxdart of the pubspec, no one error has ocurred, why this is happening? how can I upgrade this 2 libraries?

[image]

rxdart: "0.23.1"
flutter_blue: ^0.6.3+1

saulopef, 2020-01-10:
Temporarily solved by using dependency_overrides.

mciekurs2, 2020-01-15:
Same problem here but with bloc. Is there any temporary solutions I can do to fix it?

madc, 2020-01-15:
@mciekurs2 My temp. solution is getting the current master branch of flutter blue:
```
  flutter_blue:
    git:
      url: git://github.com/pauldemarco/flutter_blue.git
      ref: master
```
Not yet sure, if everything works properly tough.

madc, 2020-01-28:
Just a quick update: Everything seems to work fine from master (for now).

saulopef, 2020-02-13:
yes, using Master seems to be a great solution for now, closing this, and wait for version actualization.

closed 2020-02-13 by saulopef