pauldemarco/flutter_blue #479: iOS no issue but android can not connect

loveQingQing, 2020-01-16:
E/flutter (24076): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: TimeoutException after 0:00:10.000000: Failed to connect in time.
E/flutter (24076): #0      BluetoothDevice.connect.<anonymous closure> (package:flutter_blue/src/bluetooth_device.dart:33:9)
E/flutter (24076): #1      _rootRun (dart:async/zone.dart:1122:38)
E/flutter (24076): #2      _CustomZone.run (dart:async/zone.dart:1023:19)
E/flutter (24076): #3      _CustomZone.runGuarded (dart:async/zone.dart:925:7)
E/flutter (24076): #4      _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:965:23)
E/flutter (24076): #5      _rootRun (dart:async/zone.dart:1126:13)
E/flutter (24076): #6      _CustomZone.run (dart:async/zone.dart:1023:19)
E/flutter (24076): #7      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:949:23)
E/flutter (24076): #8      Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:23:15)
E/flutter (24076): #9      _Timer._runTimers (dart:isolate-patch/timer_impl.dart:384:19)
E/flutter (24076): #10     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:418:5)
E/flutter (24076): #11     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:174:12)
E/flutter (24076):

loveQingQing, 2020-01-16:
sometimes log this:
[ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null)
E/flutter (32468): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
E/flutter (32468): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:321:33)
E/flutter (32468): <asynchronous suspension>
E/flutter (32468): #2      BluetoothCharacteristic.write (package:flutter_blue/src/bluetooth_characteristic.dart:123:10)

pauldemarco, 2020-04-02:
What steps are you taking to see the error?

Can you share the code that’s causing it?

Feel free to create another ticket with the above.

closed 2020-04-02 by pauldemarco