pauldemarco/flutter_blue #481: Double notifications when re-connecting

MrHazee, 2020-01-17:
Hi,
Thanks for all your wonderful work @pauldemarco 
I am experiencing trouble like #317. This occurs when I reconnect to my nrf 52 DK after loosing connection. It doesnt matter if I disconnect manually by (running await device.disconnect()) or if I am forcing a disconnect by restarting the nrf 52 DK. In issue #317 the soulution seems to be:
222: gattServer.disconnect();
223: gattServer.close();  // also removes listeners! 
Can I somehow achieve this from within flutter?
Thanks /Rasmus

HipSze, 2020-01-25:
I have the same issue. When disconnect the device and connect again, it will have double notifications.

tomcatvr, 2020-01-27:
Same problem here

aitch-3, 2020-01-28:
Same problem here. 

issue reproduce:
1. connect peripheral(1).
2. disconnect peripheral(1).
3. connect other peripheral(2).
4. disconnect peripheral(2).
5. connect to the first peripheral(1) again.

the very first peripheral connected emits double notification values.

if more than 1 peripheral gets connected, (peripheral(2), peripheral(3), peripheral(4).....) the first peripheral will emit multiple notification values.

aitch-3, 2020-01-28:
> if more than 1 peripheral gets connected, (peripheral(2), peripheral(3), peripheral(4).....) the first peripheral will emit multiple notification values.

So, i checked again, the problem was on my end.

 the `setNotifyValue(false)` was never called. And the next time I try to  call `setNotifyValue(true)` then `setNotifyValue(false)` gets called. 

Since I can't use await in `dispose()` function, I had to use `then((_) {})` to stop notification.

MrHazee, 2020-01-28:
I was able to solve this (tack Per) by making sure every listener really is canceled on disconnect().

subscription = listen((){});
subscription.cancel()

The problem was on my end and not flutter_blue.

closed 2020-01-28 by MrHazee

zwells, 2020-02-03:
I'm having the same issue, can somebody post an example of their disconnect function that fixed this?

zwells, 2020-02-03:
Found my issue. I didn't set _rx.value.listen() to a StreamSubscription so that I could properly call .cancel();