pauldemarco/flutter_blue #482: Characteristic read() never returns on a  HID over GATT device.

juani15151, 2020-01-18:
Having a HID device (any bluetooth button, or a selfie stick) and using the example app, I'm trying to setup a listener for a button pressed, I understand that's done by reading/listening to the Report characteristic (0x2A4D) of the HID protocol.

Neither `await c.read()` nor `await c.setNotifyValue(true); c.value.listen(...)` work as expected, instead `await c.read()` never returns.

When reading the notification descriptor after `c.setNotifyValue(true)`, flutter_blue informs that the value could be enabled (value set to [11]).

The report characteristic appears twice, might it be a problem? That part might be related to #213 but I still have the same problem for the 'report_map characteristic (0x2A4B)' wich only appears once.

When pairing the device only using Android, the button pressed triggers the 'Volume Up' key, so the device is reporting correctly. (Also tested this with the device 'keyboard mode' disabled on android settings).

GATT Documentation:
* GATT HID service: https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Services/org.bluetooth.service.human_interface_device.xml
* GATT Report characteristic: https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.report.xml

pauldemarco, 2020-04-02:
This will be quite difficult to debug without having the peripheral next to me, but if you can supply me with all the services, characteristics, and descriptors I will try to replicate your setup.

Have you tested this working with any other Bluetooth apps, such as Nordic’s nRFConnect?

juani15151, 2020-04-03:
I'm not sure what you are requesting... the device has the [services](https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Services/org.bluetooth.service.human_interface_device.xml) and [characteristics](https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.report.xml) linked on the post.

In my case, I couldn't connect to the device with the Nordic's nRFConnect. But I'm not sure what's the problem.

Any basic bluetooth device should be an HID device, if you have a bluetooth mouse or some headphones with a button, they might implement it too. I tested it in 2 different devices before posting.

halleytoken, 2022-07-23:
> Having a HID device (any bluetooth button, or a selfie stick) and using the example app, I'm trying to setup a listener for a button pressed, I understand that's done by reading/listening to the Report characteristic (0x2A4D) of the HID protocol.
> 
> Neither `await c.read()` nor `await c.setNotifyValue(true); c.value.listen(...)` work as expected **for the report and report_map characteristics**, instead `await c.read()` never returns.
> 
> When reading the notification descriptor after `c.setNotifyValue(true)`, flutter_blue informs that the value could be enabled (value set to [11]).
> 
> Other service's characteristics work as expected.
> 
> The report characteristic appears twice, might it be a problem? That part might be related to #213 but I still have the same problem for the 'report_map characteristic (0x2A4B)' wich only appears once.
> 
> When pairing the device only using Android, the button pressed triggers the 'Volume Up' key, so the device is reporting correctly. (Also tested this with the device 'keyboard mode' disabled on android settings).
> 
> GATT Documentation:
> 
> * GATT HID service: https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Services/org.bluetooth.service.human_interface_device.xml
> * GATT Report characteristic: https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.report.xml

2 years later,I got the the same issue, and I finally knew it is due to Android security scheme call BLUETOOTH_PRIVILLEDGED.  Android will block the HID UUID R/W operation without BLUETOOTH_PRIVILLEDGED permission. while the permission is only possible for system signed APK. wondering if anyone have a new way to bypass the LIMITATION.