pauldemarco/flutter_blue #486: Android X ScanResult empty list

KevinRohn, 2020-01-24:
Hello community,

I'm using Android X and Android 9 devices for testing. 
For Android X I've a huge problem. The ScanResults are always empty. On Android 9 Devices it shows me the correct ScanResults. 

Same Project. compileSDKVersion is 29. I also tried to change the permissions from  ACCESS_COARSE_LOCATION to ACCESS_FINE_LOCATION, but no changes. 

Does someone have any idea how to fix it? 

Thank you in advance
cheers,
Kevin

tomcatvr, 2020-01-27:
Hi, there is still a bug in this version. I read it sometime ago.
Android X need ACCESS_FINE_LOCATION but it has to be put in the module permission.
My suggestion is to use compileSDKVersion is 28 till the new version

x0054, 2020-01-31:
See this pull request:

https://github.com/pauldemarco/flutter_blue/pull/383

It, along with my pull request for fixing similar issue with iOS 13.3 need to be merged in. To the author: please consider merging those 2 requests so that Flutter Blue can target the latest OS builds.

KevinRohn, 2020-02-03:
@x0054 I tried to implement the changes from your pull request, but it also ends in an empty scan result.

KevinRohn, 2020-02-11:
Any news? I'm really stuck on the issue.

mudlee, 2020-02-16:
Same here (osxygenos with android 10)

roryzh, 2020-02-19:
Same here. My app used to work on my iOS 13.3, then I tested my app on Android 10, my app gives "another scan is already in progress exception". And the scan result list is empty. My app function is scanning for a BLE device that is advertising data without connection required. My app scan for this device once per second to update data since I cannot connect to it.

So I reinstall the example app from flutter_blue, it runs without exception, but the BLE device list is always empty.

mudlee, 2020-02-20:
@pauldemarco are you still maintain this package? If not, it's OK for sure, just to know if I have to just wait or I have to either fix it or find an another package. Thanks.

roryzh, 2020-02-20:
@mudlee Please let me know if you find another package. I didn't find one. I am planning to use Swift for now.

mudlee, 2020-02-21:
> @mudlee Please let me know if you find another package. I didn't find one. I am planning to use Swift for now.

I have not found either, but I'm optimistic :P There is one, but it's still lacks something very basic, see here: https://github.com/PhilipsHue/flutter_reactive_ble/issues/12

dokinkon, 2020-03-14:
I have some problem. After struggling few hours, I found just change targetSdkVersion to 28 instead 29 can fix this issue. Hope can help you guys.

boskokg, 2020-03-15:
Note:
I added your code together with other merge request.

  flutter_blue:
    git:
      url: https://github.com/boskokg/flutter_blue.git
      ref: 0.7.1

dokinkon, 2020-03-23:
@boskokg 
I found current android plugin is not compatible with Flutter 1.12+. I try to migrate to flutter 1.12. If you are interested with this issue, merging into your fork is welcome.

https://github.com/dokinkon/flutter_blue/commit/1b828895330878ed6ccb3e6271ccf0e1a89ae7b2

For your reference
https://flutter.dev/docs/development/packages-and-plugins/plugin-api-migration

boskokg, 2020-03-23:
@dokinkon 
I use flutter 1.16.1 and this plugin works in my project (My activity still extends FlutterActivity).

Is your change compatible with both old and new approach? I would not like to cut off the people that cannot migrate to new approach because the still depend on plugins that are not migrated.

ride4sun, 2020-03-24:
> flutter_blue:
> git:
> url: https://github.com/boskokg/flutter_blue.git

@pauldemarco I am wondering if you could give others access to this repository and merge some of the old merge requests. @boskokg would you willing to do that? I don't have the background unfortunately

dokinkon, 2020-03-25:
> @dokinkon
> I use flutter 1.16.1 and this plugin works in my project (My activity still extends FlutterActivity).
>
> Is your change compatible with both old and new approach? I would not like to cut off [...]

@boskokg 
Got it. I will take a look into compatible with legacy approaches. Thanks for reminding.

pauldemarco, 2020-04-01:
@KevinRohn 
This should be resolved in the latest version 0.7.0

closed 2020-04-01 by pauldemarco

Mr-xzq, 2021-11-18:
I also meet a similar problem in Android X，flutter_blue version is 0.7.3。 

1. You can attempt set 
[image]

2. and after manual open your mobile gps location (this is important)。