pauldemarco/flutter_blue #487: Support web

masoudparsa, 2020-01-25:
Does the module support the web?

raphael-bmec-co, 2020-01-30:
I would be interested in implementing this.

These are the browsers that support it:

Chrome v. 56 and newer
Opera v. 43 and newer
Android browser v. 67
Opera mobile v. 46
Chrome for Android v. 70
Samsung Internet v 6.2

Worth it?

raphael-bmec-co, 2020-01-30:
Upon further reading it seems like you should be able to access the Web Bluetooth API directly by using the [html package](https://pub.dev/packages/html) and then accessing html.navigator.: [scanning](https://webbluetoothcg.github.io/web-bluetooth/scanning.html), [device](https://webbluetoothcg.github.io/web-bluetooth/).bluetooth etc.

dawsonc623, 2020-02-23:
@raphael-bmec-co I might be missing something, but I do not see any references to `navigator` in the Dart package you linked. That said, I still think it is worth exploring deeper for getting Bluetooth support into this package. My workplace is very interested in cross-platform Bluetooth connectivity.

raphael-bmec-co, 2020-02-24:
@dawsonc623 [This ](https://medium.com/funwithflutter/flutter-web-importing-html-and-javascript-flutter-1-9-c728ae9eaf2f) article may help clear up how to use the html package. There isn't a specific reverence to navigator but navigator forms part of the html APIs that are wrapped by the Dart package. So you can access it by including the package and "html.navigator" etc etc.

I agree that it would be great to have it all integrated in flutter_blue. Maybe @pauldemarco can give us some input on how feasible he thinks this would be? I don't have a clear picture of if/when it is possible to write packages that work across all native and web platforms.

bean5, 2020-02-27:
> Upon further reading it seems like you should be able to access the Web Bluetooth API directly by using the [html package](https://pub.dev/packages/html) and then accessing html.navigator.: [...]

So if this package leveraged `kIsWeb` from foundation.dart, it would know whether to use html.

However, I don't see any references to bluetooth in the [dart-lang/html](https://pub.dev/packages/html) package.

borisllona, 2021-03-17:
> @raphael-bmec-co I might be missing something, but I do not see any references to `navigator` in the Dart package you linked. That said, I still think it is worth exploring deeper for getting [...]

Has someone found something in that regard? Is it possible to use flutter_blue for web development?