pauldemarco/flutter_blue #489: AndroidX deprecation and protobuf error

an01f01, 2020-01-31:
I keep having this issue with AndroidX and my projects will not compile, has anyone been able to figure out a solution?

Note: C:\dev\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\anegri\dev\flutter\certificate\bmi-calculator\build\flutter_blue\generated\source\proto\debug\javalite\com\pauldemarco\flutter_blue\Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Running Gradle task 'assembleDebug'...
Running Gradle task 'assembleDebug'... Done                         7.1s
Gradle build failed to produce an .apk file. It's likely that this file was generated under
C:\Users\anegri\dev\flutter\certificate\bmi-calculator\build, but the tool couldn't find it.

boskokg, 2020-02-22:
I think you need to enable Jetifier. My graddle properties are:

android.enableR8=false
android.useAndroidX=true
android.enableJetifier=true

pauldemarco, 2020-04-01:
This should be resolved in the latest version.

closed 2020-04-01 by pauldemarco

kidfrom, 2020-07-12:
@pauldemarco confirmed in 0.7.2, the problem is not resolved.

```
Launching lib\main.dart on Android SDK built for x86 in debug mode...
Note: C:\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.7.2\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\user\Documents\GitHub\flutter-boilerplate\build\flutter_blue\generated\source\proto\debug\javalite\com\pauldemarco\flutter_blue\Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
√ Built build\app\outputs\apk\debug\app-debug.apk.
I/FlutterBluePlugin(25553): setup
Connecting to VM Service at ws://127.0.0.1:50881/gzCe4B4xrjY=/ws
D/EGL_emulation(25553): eglMakeCurrent: 0xaab85480: ver 2 0 (tinfo 0xaab835a0)
```

Luiggi13, 2020-07-30:
> @pauldemarco confirmed in 0.7.2, the problem is not resolved.
>
> ```
> Launching lib\main.dart on Android SDK built for x86 in debug mode...
> Note: [...]

Same as you, I can't compile. What can we do?

I have same as @pauldemarco 

android.enableR8=false
android.useAndroidX=true
android.enableJetifier=true

Nothing works

ln-12, 2020-08-25:
Same here, the deprecation message is still there.