pauldemarco/flutter_blue #490: Connect to a BLE Device Without Searching

IAmSarthakVerma, 2020-02-04:
Can we connect to a BLE Device without scanning if we know the name of the BLE firsthand?

Fleximex, 2020-02-18:
You can using the MAC-Address.

IAmSarthakVerma, 2020-02-18:
@Fleximex Can you please show a small sample code or just one line of how to modify device.connect().

Fleximex, 2020-02-18:
You should create the Device object yourself. Use that to connect (``device.connect()``).
The only problem is that there is no constructor since a commit made in December 2018. You can change the code to add a constructor yourself or figure out how to use the from Proto method.
https://github.com/pauldemarco/flutter_blue/issues/344

boskokg, 2020-02-19:
Also take a note that you can connect using MAC address only on Android (on iOS you cannot).

pauldemarco, 2020-04-01:
Constructing BluetoothDevices manually is not recommended as it won’t work cross platform (only works on Android since iOS doesn’t allow connection by MAC address)

The recommended approach is to scan for devices nearby, filter out the one you are looking for, then connect to it.

closed 2020-04-01 by pauldemarco