pauldemarco/flutter_blue #494: Getting a huge amount of output in terminal while notifications are activated

esaupcb, 2020-02-05:
For now I am activating notifications for 3 characteristics at a time. The values change every 10ms, so I have a lot of output in my terminal with the following message:

D/FlutterBluePlugin(12964): [onCharacteristicChanged] uuid: 2e43xxxx-4745-xxxx-6248-20627920444d
D/FlutterBluePlugin(12964): [onCharacteristicChanged] uuid: 2e43xxxx-4745-xxxx-6248-20627920444d
D/FlutterBluePlugin(12964): [onCharacteristicChanged] uuid: 2e43xxxx-4745-xxxx-6248-20627920444d

Is there a way to turn off this message?
Otherwise there is no way to use print for debugging..
Thanks in advance!

thachnb85, 2020-02-06:
You can set log level to other not debug, for example:
```
  FlutterBlue _flutterBlue = FlutterBlue.instance;
  _flutterBlue.setLogLevel(LogLevel.emergency);
```

esaupcb, 2020-02-07:
It works. Thanks a lot!

closed 2020-02-07 by esaupcb