pauldemarco/flutter_blue #497: PlatformException(connect, Peripheral not found, null)

luisvilches, 2020-02-17:
Hi,
i have the next error at method connect, only in ios, someone knows how to fix it?

PlatformException(connect, Peripheral not found, null)

sami-samcla, 2020-02-19:
I have the same problem. 

This is my source code:

```
void scanForDevices() async {
    sController = StreamController();

    //await checkConnectedDevices();

    sController.addStream(FlutterBlue.instance.scanResults);

    FlutterBlue.instance.startScan(timeout: Duration(seconds: 10));

    scanSubscription = sController.stream.listen((devices) async {
      final result = devices.where((currentDev) => currentDev.advertisementData.localName.contains((_dev as BaseDevice).psn)).toList();

      if (null == deviceToConnect && null != result && result.length > 0) {
        final dev = result.first;
        deviceToConnect = dev.device;
        print(dev.advertisementData.localName);

        try {
          await dev.device.connect();

          await checkState();

          setState(ViewState.Idle);
        } catch (e) {
          setState(ViewState.Idle);

          print(e);
        }

        //stopScanning();

        setState(ViewState.Idle);
      }
    });
  }

void stopScanning() async {
    print('stopping scan');

    await FlutterBlue.instance.stopScan();

    if (null != scanSubscription) await scanSubscription.cancel();

    if (null != sController) await sController.close();
  }
```
Works fine, but not always. For example:
- first time works fine
- second time doesn't work
- thirth time works fine
- fourth time doesn't work
- ...

boskokg, 2020-02-19:
You should probably provide the full code example (eg. where do you hold field deviceToConnect and when you clear it).

Note1 : sController.stream.listen will remain when scanning timeout happens.
Note2: commented stopScanning() should be called as soon as you detect the scanned device (before any await).

sami-samcla, 2020-02-20:
> You should probably provide the full code example (eg. where do you hold field deviceToConnect and when you clear it).
>
> Note1 : sController.stream.listen will remain when scanning timeout happens. [...]

@boskokg please find above the full code. Now I call stopScanning() method just before I detect the scanned device.
Thanks for your comments.

sami-samcla, 2020-03-02:
@boskokg did you find out any potencial problem in the full code?
Thanks in advance!!

pauldemarco, 2020-04-01:
@sami-samcla Sorry for the delay.

Are you still experiencing this issue?

sami-samcla, 2020-04-01:
Hello Paul.

Yes I'm still experiencing this issue.

Do you have any suggestion to solve it.

Many thanks for your attention.

KeenStratium, 2020-05-22:
Also experiencing this issue randomly.

Alex-Evo, 2020-05-27:
I'm experiencing a similar issue. So far, my conclusions are as follows:
- iOS keeps a list of devices connected to the phone (which is obtained by flutter_blue using the .connectedDevices attribute - mapping to iOS' "retrieveConnectedPeripheralsWithServices")
- If a device is not in that list, it is discovered by scanning, and calling .connect() on such a "freshly discovered" device always works for me
- If a device is in that list, it will not be discovered by scanning. I can still call connect on the BluetoothDevice instance I get from .connectedDevices, but now the issues begin...
- If the last discovery of the device has been more or less recent, connecting usually works
- If the last discovery of the device has been longer ago (let's say 1-2 hours), connecting does not work, and I get the "PlatformException(connect, Peripheral not found, null)"
- What's really bad is that from that point on, I can neither connect nor re-discover the device by scanning, as it appears to be still in the connectedDevices list but not working anymore
- If I turn Bluetooth off and on again at this point, it appears that the connectedDevices list gets cleared and I can re-discover the device again by scanning and also connect to it again successfully

- If I wait longer than the "critical window", I may not run into the issue at all, because the device has been removed from the connectedDevices list by iOS? I think that explains the "randomness" of the effect.

- It appears to me that different iPhones (or different iOS versions) not sure about that are affected to different extents. My own iPhone Xs running iOS 13.4.1 appears to be not affected, and devices disappear from the connectedDevices list within a few minutes after I call disconnect() in my app. One of my colleagues has an iPhone 8 (not sure about the iOS version at this point, need to double-check), which runs into the problem quite often. It appears that on his phone, devices are staying much longer in the connectedDevices list.

Hope that helps to figure out more about the issue. If anyone knows a workaround that would be highly appreciated.

Alex-Evo, 2020-05-29:
Yesterday, one of our test users reported the same issue also with an iPhone 11, so it does not appear to be limited to older hardware like the iPhone 7 Plus I mentioned in my previous post.

mbaz2, 2020-06-16:
We are also having this issue with iOS from devices 5c, iphone 11, Xs, 6+. Seldomly happens on android but is still existent.

Alex-Evo, 2020-06-16:
@pauldemarco: Any chance on looking into this topic? It is pretty detrimental to our user experience. I would also be happy to help looking into the issue (coding-wise) if you have some pointers what it could be related to.

mbaz2, 2020-06-23:
@Alex-Evo I've noticed on iOS when a device gets into this state trying to press our 'Connect' button this error will will persist and doesn't seem to resolve itself. If I transition away from the connection screen and back though it will work as intended.

darrendavidhumphrey, 2020-06-23:
Yes, me know if we can help debug this issue.  It is also pretty
detrimental to our device setup experience.  We just released 1.0 of our
app to the IOS store with the issue.

Alex-Evo, 2020-06-24:
@mbaz2: Can you share what happens code-wise in your application when you
a) transition to your connection screen
b) press your connect button
c) transition away from your connection screen

Maybe there are some pointers in there that would help us either find a work-around or identify the core of the issue.

darrendavidhumphrey, 2020-06-24:
I will try to create a minimal app that demonstrates the issue.

Huacanacha, 2020-07-15:
I have a potential workaround - see code below.

I suspect there is some caching or other state that isn't being cleaned up properly. When I force recreate the FlutterBlue.instance the problem goes away (in my particular implementation at least). I saw this error when trying to reconnect to BT after the app resumes from background state, which I was doing so I could clear the BT connection while not in active use.

My flow was:
1. Init & scan etc
2. Connect device
3. Discover services & Characteristics etc
4. Do stuff
5. Disconnect all BT devices (triggered by app entering background 'paused' state)
6. Resume app
7. Init and scan etc
=> note: after finding the correct device (and before connecting or calling discover) it still had the list of Services & Characteristics that were found previously... **this is a red flag** (these should be empty).
8. Connect device ==> **PlatformException(connect, Peripheral not found, null)**

My workaround is to add this BlowItUp method to the FluttereBlue class and call it before getting FlutterBlue.instance whenever I connect to BT. This ensures each time I try to reconnect I get a freshly instantiated FlutterBlue. 
```
  static FlutterBlue _instance = new FlutterBlue._();
  static FlutterBlue get instance => _instance;

  static BlowItUp() {
    _instance = new FlutterBlue._();
  }

```
This isn't ideal given FlutterBlue is designed as a singleton, and there may be side effects, but for now it's working for me.

Alex-Evo, 2020-07-16:
Very interesting finding @Huacanacha - sounds like something does not get properly (re-)initialized. Maybe this helps us get to the bottom of the issue...

JUREXT, 2020-08-06:
Hi,

I have forked the lib, added this workaround, and testing it on android device-works, but on iphone 4s the error ramains.
Maybe I am doing something wrong, has anyone else managed to remove this error.

Huacanacha, 2020-08-06:
@JUREXT I have used this workaround for iPhone 8 and iPhone 11 devices so as far as I know it works for iOS. Not sure if it may be an issue with older hardware or iOS versions - have you tested on a more recent iOS device?

JUREXT, 2020-08-07:
@Huacanacha I have only iphone 4s for testing currently. Will try it on newer version, but have to get a phone first.

JUREXT, 2020-08-07:
@Huacanacha I was thinking, do you maybe have a forked version of this lib, where you say it works, could you give me the link, so I can test in my test app, to see have it behaves.

Huacanacha, 2020-08-07:
@JUREXT no I didn't fork. I just fixed locally given I see this as a temporary workaround. The commit in your fork looks the same as what I have in my local flutter_blue code. I do also make sure to stopScan() and disconnect all devices before recreating the instance... something like this:

```
var flutterBlue = FlutterBlue.instance;
await flutterBlue.stopScan();
for (var device in await flutterBlue.connectedDevices) {
  await device.disconnect();
}
```

JUREXT, 2020-08-08:
@Huacanacha Thank you.

yiguozhen0510, 2020-09-11:
Hi,
PlatformException(connect, Peripheral not found, null)
i also have the  error at method connect, only in ios, anyone knows how to fix it?

Huacanacha, 2020-09-11:
> Hi,
> PlatformException(connect, Peripheral not found, null)
> i also have the error at method connect, only in ios, anyone knows how to fix it?

Did you try my workaround?

joankabello, 2020-10-19:
> > Hi,
> > PlatformException(connect, Peripheral not found, null)
> > i also have the error at method connect, only in ios, anyone knows how to fix it?
> 
> Did you try my workaround?

@Huacanacha can you provide a full example of the 2 workarounds that you mentioned ?

I used both of them and it didn't fixed the error.
Maybe i have missed smth.

aBuder, 2020-11-04:
Is any update here? Is there an wokaround without forking?

BananaNeil, 2021-02-01:
I'm also occasionally experiencing this issue with a macos app. I'd love to get an update on this!

joankabello, 2021-02-01:
https://github.com/joankabello/flutter_blue

 I made some fixes on this forked repo.
 It fixed the issue with the PlatformException.

jlubeck, 2021-03-04:
@joankabello would be nice if you made a PR with the fix into this repo. And maybe explain what the issue was?

joankabello, 2021-03-05:
Hello @jlubeck ,

here is the link to the issue that i also explain the problem and how i solved it.
https://github.com/pauldemarco/flutter_blue/issues/707#issuecomment-790470158

maddionatoplus, 2021-06-02:
Hello everybody. Except for the workarounds. Is there an official fix or solution to this problem?
thank you in advance.

zhangzhenge2, 2021-07-01:
A simple solution：   Connect the device one or more second after scanning and do not stop scan immediately；

for example:
//here  scan
 await Future.delayed(Duration(seconds: 1));
//here connect

reason: flutter_blue plugins need some time to add the scanned device to _scannedPeripherals;

This is a plug-in problem. I haven't found out the specific reason

sami-samcla, 2021-07-01:
> A simple solution： Connect the device one or more second after scanning and do not stop scan immediately；
>
> for example:
> //here scan
> await Future.delayed(Duration(seconds: 1));
> //here connect [...]

This solution doesn't work. Same behavior if I put a Future.delayed after get the ScanResult device.

AlexHartford, 2021-07-02:
The above solutions don't work for me as well. What I'm doing works flawlessly on Android but keep getting the `peripheral not found` error on iOS.

When a user connects to a device, I save the info so I can immediately connect next time they open the app. So I'm not doing any scanning, etc. that needs to have delays. I create a BluetoothDevice object myself and attempt to connect to it. Not sure where to look but any pointers someone could offer would help a lot. Again, this works perfectly fine on Android.

maddionatoplus, 2021-07-04:
Is there a way to contact the author of the project? 
This is very important for me for a project.

groothuisss, 2021-09-24:
> A simple solution： Connect the device one or more second after scanning and do not stop scan immediately；
>
> for example:
> //here scan
> await Future.delayed(Duration(seconds: 1));
> //here connect [...]

Adding `await Future.delayed(Duration(seconds: 2));` worked for me, thanks!

fabiototh, 2021-12-02:
@pauldemarco do you have any predictions on this topic?

JohnnyFreire97, 2021-12-03:
> The above solutions don't work for me as well. What I'm doing works flawlessly on Android but keep getting the `peripheral not found` error on iOS.
>
> When a user connects to a device, I save the info [...]

Tendo o mesmo problema aqui

plut9, 2021-12-12:
Any luck, @JohnnyFreire97 ?

melihkaragoz, 2021-12-30:
still ?

vaibhiarora03, 2022-01-22:
Having the same issue.

fabiototh, 2022-01-22:
I solved this: 
https://github.com/TOTHLIFECARE/flutter_blue

vaibhiarora03, 2022-01-22:
Hi @fabiototh, thanks for sharing this. What was the exact error?

fabiototh, 2022-01-22:
@vaibhiarora03 in iOS, verified only in the list of scanned devices. When try connect, return this error "Peripheral not found" because the peripheral is in the connected list.

vaibhiarora03, 2022-01-22:
@fabiototh okay. i will surely integrate it. thanks for the help

HninNweAye, 2022-03-03:
> Adding `await Future.delayed(Duration(seconds: 2));` worked for me, thanks!

Worked for me too..Thanks.

WeiKuang, 2022-03-31:
> 非常有趣的发现@Huacanacha - 听起来有些东西没有正确（重新）初始化。也许这可以帮助我们找到问题的根源......
> 
> 我还将尝试您的解决方法，看看这是否对我们的情况也有帮助。

Did you solve it? Can you share how to solve it?

BrianGaleanoSion, 2022-08-27:
i solve it, modify the file flutter_blue/ios/Classes/FlutterBluePlugin.m

In method stopScan, i added new delay before get method stop

` else if([@"stopScan" isEqualToString:call.method]) {
      
      double delayInSeconds = 0.2; // set the time 
      dispatch_time_t popTime = dispatch_time(DISPATCH_TIME_NOW, delayInSeconds * NSEC_PER_SEC);
      dispatch_after(popTime, dispatch_get_main_queue(), ^(void){
        // do something here
          [self->_centralManager stopScan];
      });
    
    result(nil);
  } `

briangaleano, 2022-08-27:
i solve it, modify the file flutter_blue/ios/Classes/FlutterBluePlugin.m
In method stopScan, i added new delay before get method stop
` else if([@"stopScan" isEqualToString:call.method]) {
  double delayInSeconds = 0.2; // set the time 
  dispatch_time_t popTime = dispatch_time(DISPATCH_TIME_NOW, delayInSeconds * NSEC_PER_SEC);
  dispatch_after(popTime, dispatch_get_main_queue(), ^(void){
    // do something here
      [self->_centralManager stopScan];
  });

result(nil);
} `

briangaleano, 2022-08-27:
i solve it, modify the file flutter_blue/ios/Classes/FlutterBluePlugin.m
In method stopScan, i added new delay before get method stop

`else if([@"stopScan" isEqualToString:call.method]) {
  double delayInSeconds = 0.2; // set the time 
  dispatch_time_t popTime = dispatch_time(DISPATCH_TIME_NOW, delayInSeconds * NSEC_PER_SEC);
  dispatch_after(popTime, dispatch_get_main_queue(), ^(void){
    // do something here
      [self->_centralManager stopScan];
  });

result(nil);
}`

mevluttosun, 2022-09-13:
I've still same issue. Maybe I am using proto to connect device. In android I have no issue but in my test device IOS 12.5.5, ( I hadn't try with another device) always giving same PlatformException(connect, Peripheral not found, null, null)
Here is my proto extention 
`extension BluetoothDeviceExtension on BluetoothDeviceModel {
  BluetoothDevice toBluetoothDevice() {
    FlutterBluePlus.instance.scan();
    proto.BluetoothDevice p = proto.BluetoothDevice.create();
    p.name = name;
    p.type = proto.BluetoothDevice_Type.LE;
    p.remoteId = id;
    return BluetoothDevice.fromProto(p);
  }
}
`
Here is my connect method
`  Future<void> connect(bool autoConnect) async {
    log.fine('connecting');
    emit(const BluetoothModelDeviceState.connecting());
    final bluetoothDevice = device.toBluetoothDevice();
    await bluetoothDevice
        .connect(
          autoConnect: autoConnect,
        )
        .onError((error, stackTrace) =>
            log.warning('Connection Error ${error.toString()}'));
  }
`

Nicolas-Sugranes, 2022-09-19:
I managed to get it work by doing this :

First, I scan the results to search the one I am looking for. One I found it, I call the stopScan() method, I save the founded device in a global variable and I call a second method to make the needed operations with the device. Then, when I call my  first method a new time, I use the saved variable to connect to the device without searching for it a second time.

Here is an exemple :

`BluetoothDevice? foundedDevice = null
FlutterBlue flutterBlue = FlutterBlue.instance;

void startBleProcess() async {
    if (foundedDevice != null) {
        makeOperations(foundedDevice);

        return null;
    }

    flutterBlue!.startScan(timeout: Duration(seconds: 10));

      flutterBlue!.scanResults.listen((events) async {
        for (ScanResult result in events) {
          if (isCorrectDevice(bleMacAddress) && (await flutterBlue!.connectedDevices).length == 0) {
            await flutterBlue!.stopScan();
          
            foundedDevice = result.device;
            makeOperations(result.device);
          }
        }
      });
}

void makeOperations() async {
      await device.connect(timeout: const Duration(seconds: 10));

     // then, do your stuffs here
}
`

I am a beginner in Flutter development, so idk if this is a "clean" solution, but I hope it will work for you aswell !

shrenikshah-wama, 2023-03-30:
Did anyone find any solution to this?
I'm still facing the same issue. 

For iOS, First time device discovery, service & characteristics discovery, read/write operations and disconnection works perfect. But when I try to reconnect, it throws 

` ERROR : PlatformException(connect, Peripheral not found, null, null)`

error and unable to proceed ahead

anakin-jin, 2023-04-28:
> A simple solution： Connect the device one or more second after scanning and do not stop scan immediately；
>
> for example: //here scan await Future.delayed(Duration(seconds: 1)); //here connect [...]

this solution works for me.
my envi: iPhone XR ios 16.2/ flutter 3.7.11 stable/flutter_blue: ^0.8.0

JastySaran, 2023-11-10:
There are few issues with Flutter Blue package. Instead use flutter_blue_plus package for bluetooth operations.

PlatformException(connect, Peripheral not found, null) is an exceptions thrown by flutter package when ever it got an exceptions from the native operating system.

The exceptions often raise due to storage of previously connected device information and it wont be nil. To make it work you need to clear the instance of already existing flutter blue instance and initiate a new instance to perform bluetooth operations.

To Overcome the issue you can use the latest version of flutter_blue_plus: ^1.28.7.

```
// Initially Start scanning
await FlutterBluePlus.startScan();

// Setup Listener for scan results after scanning
var subscription = FlutterBluePlus.scanResults.listen((results) {
        if (results.isNotEmpty) {
            // the most recently found device
            ScanResult r = results.last; 
            print('${r.device.remoteId}:"${r.advertisementData.advName}" 
            found!');
            // Stop scanning
            await FlutterBluePlus.stopScan();
            connectBleDevice(r.device);
          }
       },
       onError(e) => print(e);
    );

connectBleDevice(BluetoothDevice device) async {
    // Connect to the device
    await device.connect();
}

disconnectBleDevice(BluetoothDevice device) async {
     // Disconnect from device
    await device.disconnect();
}
```
Once after successful connect of Ble device you can perform disconnect and again try connect to it using the above mentioned steps. There will no issues in the new version of the flutter_blue_plus: ^1.28.7 package as they are removing knownServices & lastValues on reconnection instead of disconnection.

https://pub.dev/packages/flutter_blue_plus/changelog

iStarEternal, 2024-01-19:
This issue only happens on iOS. Bluetooth is not connected because the Bluetooth message is cached by flutter.
In `flutter_blue.dart` code, _scanResults uses BehaviorSubject. The PublishSubject should be used correctly here.
```flutter
BehaviorSubject<List<ScanResult>> _scanResults = BehaviorSubject.seeded([]);
```
But that's fine, even if BehaviorSubject is the one you used, it simply skips his cache, and the problem is solved.
```flutter
FlutterBlue.instance.startScan();
_scanResultSub = FlutterBlue.instance.scanResults.skip(1).listen((results) {
  // ... do somethings
});
```
Use skip(1) to skip the cache