pauldemarco/flutter_blue #500: iOS/Android: reported value of ATT mtu sometimes includes or excludes the ATT header (3 bytes).

nmbus, 2020-02-21:
Hi,

Upon connecting, the API reports an ATT MTU of 20 bytes on both platform, which is the application data MTU and not the default ATT MTU (23 bytes).
That could be an implementation choice, however the reported values upon MTU changes do not necessarily follow that rule. For example on iOS, the ATT MTU on recent iOS versions is negotiated at 185, Flutter Blue reports 182 (so data MTU), while on Android it always gets maxed out at 512 (on some phones 517), and Flutter Blue reports 512 (so the ATT MTU !). Would be nice to make it consistent so that it is always the data MTU or the ATT MTU being reported.

pauldemarco, 2020-03-31:
Nice find and writeup @nmbus

savar, 2020-08-29:
This bit me as well, wouldn't it be good to at least document that behavior for the time being until being fixed?