pauldemarco/flutter_blue #502: Manufacture data missing in Example App?  Where is the raw Advertisement Data found?

robomike, 2020-02-26:
Hi All, maybe I am just confused.  Here are screen shots of my device in nordic RF and flutter blue , how is the raw data presented in flutter blue ? 
[image]
[image]
[image]
[image]

_Originally posted by @robomike in https://github.com/pauldemarco/flutter_blue/issues/102#issuecomment-591079447_

zwells, 2020-03-09:
You can get this using device.advertisementData.manufacturerData.toString()

pauldemarco, 2020-03-31:
Hi @robomike,

As @zwells mentioned, what you are showing in the Flutter Blue screenshot is characteristic data, and in the nRF screenshot Advertisement Data is shown.

You can retrieve the advertisement data from the ScanResult with `scanResult.advertisementData`.

closed 2020-03-31 by pauldemarco