pauldemarco/flutter_blue #504: Flutter Blue minimum IOS version is 8 instead of 10.

navaronbracke, 2020-03-01:
When trying to run my Flutter app on iOS, I'm getting a build error.
The error states that the minimum supported version for the plugin is iOS 8, even though the Bluetooth API used is only available from iOS 10 onwards. Hence the building of the ``flutter_blue.h`` Header file fails. Thus killing my entire build.

```
Launching lib/main.dart on iPhone 11 Pro Max in debug mode...
 
Running pod install...                                              2,0s
Running Xcode build...                                                  
                                                   
 ├─Assembling Flutter resources...                          16,9s
 └─Compiling, linking and signing...                         1,9s
Xcode build done.                                           24,3s
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **

Xcode's output:
↳
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m
    :546:49: warning: 'CBManagerState' is only available on iOS 10.0 or newer [-Wunguarded-availability]
    - (ProtosBluetoothState*)toBluetoothStateProto:(CBManagerState)state {
                                                    ^
    In module 'CoreBluetooth' imported from
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.h
    :6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator13.2.sdk/System/Library/Fr
    ameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note: 'CBManagerState' has been marked as being introduced in iOS 10.0
    here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m
    :546:1: note: annotate 'toBluetoothStateProto:' with an availability attribute to silence this warning
    - (ProtosBluetoothState*)toBluetoothStateProto:(CBManagerState)state {
    ^
                                                                          API_AVAILABLE(ios(10.0))
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m
    :74:37: warning: 'CBManagerStateUnsupported' is only available on iOS 10.0 or newer [-Wunguarded-availability]
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.h
    :6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator13.2.sdk/System/Library/Fr
    ameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note: 'CBManagerState' has been marked as being introduced in iOS 10.0
    here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m
    :74:37: note: enclose 'CBManagerStateUnsupported' in an @available check to silence this warning
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m
    :74:95: warning: 'CBManagerStateUnknown' is only available on iOS 10.0 or newer [-Wunguarded-availability]
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                                                                                  ^~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.h
    :6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator13.2.sdk/System/Library/Fr
    ameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note: 'CBManagerState' has been marked as being introduced in iOS 10.0
    here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m
    :74:95: note: enclose 'CBManagerStateUnknown' in an @available check to silence this warning
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                                                                                  ^~~~~~~~~~~~~~~~~~~~~
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m
    :80:37: warning: 'CBManagerStatePoweredOn' is only available on iOS 10.0 or newer [-Wunguarded-availability]
        if(self.centralManager.state == CBManagerStatePoweredOn) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.h
    :6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator13.2.sdk/System/Library/Fr
    ameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note: 'CBManagerState' has been marked as being introduced in iOS 10.0
    here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/navaronbracke/Documents/Flutter/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m
    :80:37: note: enclose 'CBManagerStatePoweredOn' in an @available check to silence this warning
        if(self.centralManager.state == CBManagerStatePoweredOn) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~
    4 warnings generated.
    While building module 'flutter_blue' imported from
    /Users/navaronbracke/Documents/WeForza/weforza/weforza/ios/Runner/GeneratedPluginRegistrant.m:22:
    In file included from <module-includes>:1:
    /Users/navaronbracke/Documents/WeForza/weforza/weforza/build/ios/Debug-iphonesimulator/flutter_blue/flutter_blue.framework/Headers/flu
    tter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not found
    #import "FlutterBluePlugin.h"
            ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    /Users/navaronbracke/Documents/WeForza/weforza/weforza/ios/Runner/GeneratedPluginRegistrant.m:22:9: fatal error: could not build
    module 'flutter_blue'
    @import flutter_blue;
     ~~~~~~~^~~~~~~~~~~~
    2 errors generated.
    note: Using new build system
    note: Planning build
    note: Constructing build description
    warning: The use of Swift 3 @objc inference in Swift 4 mode is deprecated. Please address deprecated @objc inference warnings, test
    your code with “Use of deprecated Swift 3 @objc inference” logging enabled, and then disable inference by changing the "Swift 3 @objc
    Inference" build setting to "Default" for the "Runner" target. (in target 'Runner' from project 'Runner')

Could not build the application for the simulator.
Error launching application on iPhone 11 Pro Max.
```

The Swift 3 deprecation issue is probably for another ticket.

I'd suggest bumping the minimum supported version to iOS 10. No clue why it sits at the unsupported version 8 though.

nmbus, 2020-03-03:
Your error does not stem from the fact that the BLE APIs aren't available in iOS 8 (it's only warnings), but from the fact you are probably using "use_frameworks!" in your Podfile to accommodate for Swift plugins. The latter usually breaks header mappings defined in podspecs, to fix it comment this line in flutter_blue's podspec:
`ss.header_mappings_dir = '.'` (line 24)

navaronbracke, 2020-03-03:
Well I checked what's actually in the Podfile for flutter_blue. Turns out this little piece probably kills my header generation:

```
target 'Runner' do
  use_frameworks!
  use_modular_headers!
```

I'll go ahead and apply the suggestion you gave me. Will report back when it works.

navaronbracke, 2020-03-03:
Removing said line doesn't help. Once removed, the line comes back during the build step. Hence it crashes again.

navaronbracke, 2020-03-03:
It turns out that even after running flutter pub get a million times (figuratively speaking of course), the header file ``FlutterBluePlugin.h`` never exists in my filesystem. Obviously if it doesn't exist, this error happens. Any idea why I might not have this header file? FYI I just tested it with a new Flutter project. I ran ``flutter create testapp`` and added ``flutter_blue`` as a dependency. Build fails in exactly the same situation. It seems as if ``pub get`` never fetches the header file.

closed 2020-03-04 by navaronbracke

juliansteenbakker, 2020-03-12:
> Your error does not stem from the fact that the BLE APIs aren't available in iOS 8 (it's only warnings), but from the fact you are probably using "use_frameworks!" in your Podfile to accommodate for [...]

This didn't fix the problem for me. Instead, i removed `use_frameworks!` from the Podspec.

mirajago, 2020-03-12:
Neither #ss.header_mappings_dir = '.' nor #use_frameworks! fixed it for me. Still not able to build for ios. Any ideas?

Arkangel12, 2020-04-08:
> header mappings defined in podspecs

this fix the error its not sending and error using use_frameworks! in podspec but the app closes on app init did you know why?

nmbus, 2020-04-18:
> > header mappings defined in podspecs
> 
> this fix the error its not sending and error using use_frameworks! in podspec but the app closes on app init did you know why?

Most likely missing permissions, see the warning on Apple's developer documentation https://developer.apple.com/documentation/corebluetooth?language=objc

Mycircle, 2020-08-06:
> Your suggestion fixed it, thanks a lot. Closing the issue.
Hello, I met the same problem with you.
 I comment this line 'ss.header_mappings_dir = ".' in flutter_blue‘s’ podspec file according to the above steps, but I got an error after I executed the flutter run, "could not build module flutter_blue"
Thank you for any Suggestions