pauldemarco/flutter_blue #506: [Question] Maintain connection when device is locked

tdnghia98, 2020-03-01:
Hi,
Is it possible to maintain the connection and send data when the telephone is locked? I'd like to use the lib with a connected bike via bluetooth and I need to have data transmitted when the user locked their phone
Thanks.

nmbus, 2020-03-03:
Yes, while the connection is maintained in any case, you need to explicitly allow the background BLE operations on iOS, using the "UIBackgroundModes" list in your plist file, with either or both of those keys:
`bluetooth-central` If you are using the bluetooth central (aka master) mode.
`bluetooth-peripheral` If you are using the bluetooth peripheral (aka slave) mode.

tdnghia98, 2020-03-03:
Thanks @nmbus

closed 2020-03-03 by tdnghia98