pauldemarco/flutter_blue #509: Service Discovery does not terminate.

AtomicNicos, 2020-03-04:
I had an app that worked a week ago and this week hangs entirely on `device.discoverServices()` : The method never exits, but it  used to. Assigning a `timeout` value only makes the method crash.

```dart
  static final String bluetoothLeCc254xServiceUUID = "0000ffe0-0000-1000-8000-00805f9b34fb";
  static final String bluetoothLeCc254xReadUUID = "0000ffe1-0000-1000-8000-00805f9b34fb";

  Future<bool> setupCharacteristics() async {
    List<BluetoothService> services = [];
    List<BluetoothCharacteristic> characteristics = [];
    bool success = false;

    if (this._device != null) {
        services = await this._device.services.first;
        services = services.where((s) => s.uuid.toString() == bluetoothLeCc254xServiceUUID).toList();
        
        if (services.length > 0) {
          characteristics = services[0].characteristics;
          characteristics = characteristics.where((c) => c.uuid.toString() == bluetoothLeCc254xReadUUID).toList();
          if (characteristics.length > 0) {
            if (!characteristics[0].isNotifying)
              await characteristics[0].setNotifyValue(true);
            _characteristic = characteristics[0];
            success = true;
          }
        }
    }
    return success;
  }

  Future<bool> setupListener() async {
    if (_characteristic != null) {
      _characteristic.value.transform(
        new StreamTransformer.fromHandlers(
          handleData: (List<int> list, EventSink<List<int>> sink) {
            if (list.isNotEmpty)
              sink.add(list);
            else
              sink.add([]);
          }
        )
      ).listen((data) => DataHandler().addData(data));
      return true;
    } else 
      return false;
  }

  /// Makes the phone connect to the [BluetoothDevice] and initiate [BluetoothService] discovery.
  Future<void> connect() => this._device?.let(
    (it) {
      asyncConnection();
      return;
    }
  );

  asyncConnection() async {
    await this._device.connect();
    if (Platform.isAndroid)
      await this._device.requestMtu(517);
    await this._device.discoverServices();
    await setupCharacteristics();
    await setupListener();
  }
```

pauldemarco, 2020-03-31:
What is the crash error?
What phone are you using?
Please share your `flutter doctor -v`.
Can you verify this is still happening with the latest version 0.7.0?

ride4sun, 2020-03-31:
I can confirm I did see the same behavior without any exception or error message.
This happened when I was just scanning devices without being connected to it. Once I connect I could use that call  `device.discoverServices()` just fine.
```
jay@mini Projects % flutter doctor -v
[✓] Flutter (Channel beta, v1.15.17, on Mac OS X 10.15.3 19D76, locale en-US)
    • Flutter version 1.15.17 at /Users/jay/development/flutter
    • Framework revision 2294d75bfa (4 weeks ago), 2020-03-07 00:28:38 +0900
    • Engine revision 5aff311948
    • Dart version 2.8.0 (build 2.8.0-dev.12.0 9983424a3c)

 
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at /Users/jay/Library/Android/sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-29, build-tools 29.0.2
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_212-release-1586-b4-5784211)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 11.3.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.3.1, Build version 11C504
    • CocoaPods version 1.9.0

[✓] Android Studio (version 3.6)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 44.0.2
    • Dart plugin version 192.7761
    • Java version OpenJDK Runtime Environment (build 1.8.0_212-release-1586-b4-5784211)

[✓] VS Code (version 1.43.0)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.8.1

[✓] Connected device (2 available)
    • Jay’s iPhone • bc987fccd61b08a374d7fd1329958f428155b60b • ios • iOS 13.3.1
    • iPhone 8     • 2878EA5A-D561-47E4-969E-B231E23455C2     • ios • com.apple.CoreSimulator.SimRuntime.iOS-13-3 (simulator)

```

ride4sun, 2020-04-02:
I verified this with your the latest git @pauldemarco 
The problem still exist I don't see an exception or a error log entry in XCODE.
And it behaves that way on devices which are not connected when I call 
` List<ios.BluetoothService> deviceService = await device.discoverServices();`

If I set a breakpoint on the next line the breakpoint never hits.

ride4sun, 2020-04-07:
@pauldemarco is there any insides you can share with people waiting for this. Any ETA?

ride4sun, 2020-04-09:
@pauldemarco I think that can be closed

closed 2020-04-09 by pauldemarco

AtomicNicos, 2020-04-29:
Hi, this issue is still open in my case (sorry for tardiness).

The device gets stuck on `BluetoothConnectionStatus.BTSTATUS_DISCOVERING_SERVICES`. IT works on older phones.

Flutter doctor output :
```
[√] Flutter (Channel master, 1.18.0-9.0.pre.58, on Microsoft Windows [Version 10.0.18362.778], locale en-CH)
    • Flutter version 1.18.0-9.0.pre.58 at C:\src\flutter
    • Framework revision 9d106bdead (2 hours ago), 2020-04-28 23:09:50 -0700
    • Engine revision bd5234780e
    • Dart version 2.9.0 (build 2.9.0-3.0.dev 726d3c7725)

[√] Android toolchain - develop for Android devices (Android SDK version 29.0.3)
    • Android SDK at C:\Users\Atomic\AppData\Local\Android\sdk
    • Platform android-29, build-tools 29.0.3
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b03)
    • All Android licenses accepted.

[√] Android Studio (version 3.5)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin version 43.0.1
    • Dart plugin version 191.8593
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b03)

[√] IntelliJ IDEA Ultimate Edition (version 2019.3)
    • IntelliJ at C:\Program Files\JetBrains\IntelliJ IDEA 2019.2
    • Flutter plugin version 43.0.3
    • Dart plugin version 193.6015.53

[√] VS Code (version 1.44.2)
    • VS Code at C:\Users\Atomic\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.9.1

[√] Connected device (1 available)
    • Redmi Note 8 • 12fdcd54 • android-arm64 • Android 9 (API 28)
```

AtomicNicos, 2020-04-29:
Here is the debug stack after 5 minutes : 
````
D/BluetoothAdapter(23006): isLeEnabled(): ON
D/BluetoothLeScanner(23006): could not find callback wrapper
D/BluetoothManager(23006): getConnectedDevices
D/BluetoothGatt(23006): connect() - device: 19:08:10:17:17:0A, auto: true
D/BluetoothGatt(23006): registerApp()
D/BluetoothGatt(23006): registerApp() - UUID=7db13744-b9fb-497c-93ad-e51f732eaf67
D/BluetoothGatt(23006): onClientRegistered() - status=0 clientIf=7
D/BluetoothManager(23006): getConnectionState()
D/BluetoothManager(23006): getConnectedDevices
D/BluetoothGatt(23006): onClientConnectionState() - status=0 clientIf=7 device=19:08:10:17:17:0A
D/BluetoothGatt(23006): configureMTU() - device: 19:08:10:17:17:0A mtu: 517
D/BluetoothGatt(23006): discoverServices() - device: 19:08:10:17:17:0A
D/BluetoothGatt(23006): onConnectionUpdated() - Device=19:08:10:17:17:0A interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(23006): onConfigureMTU() - Device=19:08:10:17:17:0A mtu=131 status=0
D/BluetoothGatt(23006): onConnectionUpdated() - Device=19:08:10:17:17:0A interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt(23006): onConnectionUpdated() - Device=19:08:10:17:17:0A interval=12 latency=0 timeout=600 status=0
```

fgorina, 2021-06-01:
I had the same problem. 
I had a 
      

await device.connect();
await printDevice.requestMtu(128);

And it hanged forever in 

List<BluetoothService> services = await device.discoverServices();

Just removing the requestMtu works fine.

SomeoneIsWorking, 2022-08-14:
> Just removing the requestMtu works fine.

I have no `requestMtu` but my app hangs forever on `discoverServices` occasionally, I could only pin-point it after adding `.timeout`s to nearly every `Future`