pauldemarco/flutter_blue #510: Is this repository dead?

shinayser, 2020-03-04:
I've seem that the latest update on pub site was on october 2019 and here in git it has about 40+ pull requests. Most of them with no response from maintainers.

Is that repository dead?

If yes, can I suggest to transfer it's ownership?

juliansteenbakker, 2020-03-05:
+1, it seems like a good library but it would be ashame if it just died out like this.

ride4sun, 2020-03-09:
@pauldemarco I would also like to know the status of this lib because my project depends on it. Is there a active branch somewhere which has the 41 pull-requests applied to it.

juliansteenbakker, 2020-03-11:
This is @pauldemarco response on discord:

> Since this is not a paid endeavor for me, and it is currently stable for the products I’m involved in, I have to work on it in cycles. I do apologize for that and hope you all understand.
The next order of business is upgrading the plugin to be a federated plugin. Although I was waiting on the final structure to stabilize at Flutter official, it looks like they’re beginning to settle on a final template.

boskokg, 2020-03-15:
Note:
I merged several merge request to created fork (you can use it):

  flutter_blue:
    git:
      url: https://github.com/boskokg/flutter_blue.git
      ref: 0.7.1

ride4sun, 2020-03-25:
@boskokg instead of doing this in your own branch it would be great if we could reactivate this package and fix some of the overdue defects.

 @pauldemarco could u not create a branch for your company and have somebody like @boskokg take co-ownership of this package to get going of the most pressing issues here. I understand that you can not work on this as your main job but it seems like this package has quite a few followers and many of us are dependent on a support model for this package which at least gets the main Issues fixed ina timely manner. 

@boskokg would you be willing to take some ownership? Are there any other takers who have some experience with this package. Unfortunately, I don't have much experience in Bluetooth yet but if I can help with a subtask I am all ears!

boskokg, 2020-03-26:
@ride4sun My branch has the merge request waiting...
I am willing to fix the critical issues.
I am not sure if it is legal to create a new package if the maintainer stops the support?

ride4sun, 2020-03-26:
> @ride4sun My branch has the merge request waiting...
> I am willing to fix the critical issues.
> I am not sure if it is legal to create a new package if the maintainer stops the support?

@boskokg  I think it is legal as long we comply with the open-source license which we do. In a way, you are already doing that with your branch, which I highly appreciate.
I would prefer you to get co-ownership because this package has already quite a few followers.
If Paul does not make you co-owner than would you publish the package in the store? Maybe with a similar name. flutter_blue_nextgen?

@pauldemarco hey Paul it would be great if you could react to this. I think if you don't like the direction this goes you still could branch for your company/or yourself this package later.   
Please react to this one way or the other. I appreciate the work you have done but many people like to move on to get this package working again.

boskokg, 2020-03-26:
Lets then wait for one day for @pauldemarco to answer. If there is no answer, I will create a new package.

ride4sun, 2020-03-27:
> Lets then wait for one day for @pauldemarco to answer. If there is no answer, I will create a new package.

 I think we should wait a bit longer like three days. That makes it Monday to be nice. What do u think?

juliansteenbakker, 2020-03-27:
Thats some good initiative you guys are taking! If we look at the license, it's pretty clear that it's allowed to modify and redistribute the code, as long as we keep the copyright notice in all the files and fullfill the following list of conditions:

> Redistribution and use in source and binary forms, with or without
> modification, are permitted provided that the following conditions are
> met:
> 
>    * Redistributions of source code must retain the above copyright
> notice, this list of conditions and the following disclaimer.
>    * Redistributions in binary form must reproduce the above
> copyright notice, this list of conditions and the following disclaimer
> in the documentation and/or other materials provided with the
> distribution.
>    * Neither the name of Buffalo PC Inc. nor the names of its
> contributors may be used to endorse or promote products derived from
> this software without specific prior written permission.

I'm happy to volunteer as contributor. It's a good idea to apply a structured branching strategy so development can always continue. Also, all the current issues should be duplicated or something similar, so that we still keep track of the current issue's. Keep me posted 🙌

pauldemarco, 2020-03-27:
Hi all,

Please feel free to maintain any separate branches or forks that work for your use case and/or devices.

@boskokg I’ll have a look at your branch this week.

Thanks.

ride4sun, 2020-03-27:
@pauldemarco thanks for the answer. 

@boskokg I think we should go ahead of creating a new package with a slight change in the package name to point out the similarity. 

@juliansteenbakker thx for the input I think if we take it as is we should be fine and not violate any points u made, agree?

@boskokg @juliansteenbakker  I don't think we should replicate all the open defects because its hard to find out what is most pressing and what is just stale. I would rather let people file new ones and than address them as needed. There is probably a hand full of blocking defects we should address to get going and take over. I also think this lib should build and run with the flutter stable to have a good out of the box experience.  Also, the example should just work with the stable release out of the box. And we could maintain a dev branch for ongoing work. Like I said before - I am not a Bluetooth expert but I am consuming the lib and can help out as needed.

boskokg, 2020-03-27:
@pauldemarco thanks. If the merge request is good, you should close the other merge requests that are in this this merge request...

boskokg, 2020-03-27:
@pauldemarco do you plan to be more active on this package? There are issues/merge request to be closed / reviewed...
If you don't have time for that, I will go with the new package as my project depends on this...

For example, people are blocked by this plugin if using new package Android API...

ride4sun, 2020-03-31:
@boskokg I think we should go ahead and create the package and keep an eye out to merge fixes to our packages as well for now. Seems like communication is not working ...

ride4sun, 2020-03-31:
I am in general not very happy with flutters Bluetooth support. Maybe you guys want to give your 2 Cents as well. I filed an Issue.

https://github.com/flutter/flutter/issues/53493

juliansteenbakker, 2020-03-31:
Since i have a bit of a hurry in getting some issues fixed, i created a fork named flutter_ble_central, based on flutter_blue and the fixed merged by @boskokg in his fork. I added @ride4sun and @boskokg as collaborator, and added an develop branch so that we can seperate the nightly builds from stable releases. I added a gitter page to make communication a little easier. I will setup some CI/CD so that we can have regular updates, and i will be making a package for pub.dev.

pauldemarco, 2020-03-31:
Hi @boskokg,
Thanks for collating the list of merges, I was able to fold in what I could.

I would be cautious of the remaining commits, for instance:
1. The background process code for Android --Flutter has recently finalized a newer version of the embedder for Android (v2).  Proper handling of Bluetooth in the background should support v2 alongside an Android Service.  I published a version today that includes v2 support, feel free to check it out.

2. I still could not verify the iOS "build improvements" to work 100% on all of my test build environments.

@ride4sun @juliansteenbakker 
As it seems you've created another channel of communication, I feel it should be okay to close this issue? If not, please re-open, or reach me in the discord channel.

Thanks.

closed 2020-03-31 by pauldemarco

Ahmadre, 2020-10-26:
> Note:
> I merged several merge request to created fork (you can use it):
> 
> flutter_blue:
> git:
> url: https://github.com/boskokg/flutter_blue.git
> ref: 0.7.1

Error while checking out your package: rxDart collides with cross_connectivity. Update your rxDart to: 0.24.0.

StefaniOSApps, 2021-08-07:
reopen? ツ @pauldemarco