pauldemarco/flutter_blue #511: [onConnectionStateChange] on [onCharacteristicWrite] - disconnect

PhilColbert, 2020-03-04:
Anyone any ideas.

Random Chinese phone ( redmi 6)

On some writes the phone disconnects and reconnects immediately ( even with autoconnect to false ) 

D/FlutterBluePlugin(30924): mDevices size: 1
D/BluetoothManager(30924): getConnectionState()
D/BluetoothManager(30924): getConnectedDevices
D/FlutterBluePlugin(30924): [onCharacteristicWrite] uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e status: 133
D/BluetoothGatt(30924): onClientConnectionState() - status=8 clientIf=5 device=30:AE:A4:7B:76:5A
D/FlutterBluePlugin(30924): [onConnectionStateChange] status: 8 newState: 0
E/flutter (30924): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: Exception: Failed to write the characteristic

pauldemarco, 2020-03-31:
What is the peripheral you are communicating with?

jingramwright, 2020-05-22:
I'm also having this issue. 

In my instance it's while attempting to write to a characteristic on an ESP32. However, it only occurs when the MTU has been set larger than 20, and the amount of bytes passed into the write function is also larger than 20. If the MTU is 20 then any amount of bytes can be passed into the write function without this error occuring.

Any help would be much appreciated. If you want more information I'm happy to provide any you need.

Thanks!

BananaHemic, 2021-03-05:
I'm also having an issue with a status 8 disconnect with rapid reconnect. This is with an HM-10 module.
Currently, I think that this is a bug with the peripheral itself, and not this library

Yaxit, 2021-08-02:
I'm also having this problem. My device is an STM BlueNRG-MS module. 

`D/BluetoothManager(11313): getConnectedDevices
17
D/FlutterBluePlugin(11313): [onCharacteristicChanged] uuid: 00000000-0001-11e1-9ab4-0002a5d5c51b
D/FlutterBluePlugin(11313): [onCharacteristicRead] uuid: 00000000-0001-11e1-9ab4-0002a5d5c51b status: 133
D/BluetoothGatt(11313): onClientConnectionState() - status=22 clientIf=9 device=C0:7D:2D:39:41:30
D/FlutterBluePlugin(11313): [onConnectionStateChange] status: 22 newState: 0
D/BluetoothGatt(11313): onClientConnectionState() - status=0 clientIf=9 device=C0:7D:2D:39:41:30
D/FlutterBluePlugin(11313): [onConnectionStateChange] status: 0 newState: 2
D/BluetoothGatt(11313): onConnectionUpdated() - Device=C0:7D:2D:39:41:30 interval=6 latency=0 timeout=500 status=0
D/BluetoothManager(11313): getConnectedDevices
D/FlutterBluePlugin(11313): mDevices size: 1
`