pauldemarco/flutter_blue #512: Is this just for BLE?

dawsonc623, 2020-03-09:
Forgive my ignorance, as I am just starting out learning Bluetooth. Is this tool only BLE-compatible, or does it handle classic Bluetooth as well?

juliansteenbakker, 2020-03-10:
This is only for BLE related activities, not classic Bluetooth.

ConProgramming, 2020-03-19:
> This is only for BLE related activities, not classic Bluetooth.

Know of any good classic Bluetooth libraries?

juliansteenbakker, 2020-03-23:
> > This is only for BLE related activities, not classic Bluetooth.
> 
> Know of any good classic Bluetooth libraries?

I would suggest searching on [pub.dev](https://pub.dev/) for all kinds of packages related to flutter. [Take a look at these search results.](https://pub.dev/packages?q=flutter+bluetooth)

pauldemarco, 2020-03-31:
@dawsonc623 At the present time, this library is focused on Bluetooth Low Energy.

Thanks.

closed 2020-03-31 by pauldemarco