pauldemarco/flutter_blue #514: 'FlutterBluePlugin.h' file not found when compile iOS App

ipconfiger, 2020-03-11:
`dependencies:
  flutter_bloc: ^3.2.0`

than 

`$ flutter pub get`

than

`import 'package:flutter_bloc/flutter_bloc.dart';`

than compile, get error:

`/Users/alex/testspace/pinyin/build/ios/Debug-iphonesimulator/flutter_blue/flutter_blue.framework/Headers/flutter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not found
    #import "FlutterBluePlugin.h"
            ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    /Users/alex/testspace/pinyin/ios/Runner/GeneratedPluginRegistrant.m:10:9: fatal error: could not build module 'flutter_blue'
    @import flutter_blue;
     ~~~~~~~^~~~~~~~~~~~
    2 errors generated.
    note: Using new build system
    note: Planning build
    note: Constructing build description`

nmbus, 2020-03-11:
See #504

pauldemarco, 2020-04-01:
Hi @ipconfiger 
I just commited a change that I confirmed working for the example app and some swift projects. Please try with the latest version of flutter by changing your pubspec to target the master branch, like so, and let me know the results.
```
dependencies:
  flutter_blue:
    git: git://github.com/pauldemarco/flutter_blue.git
```

gib40f, 2020-04-03:
On a new and clean Flutter/Swift project I added the dependency as

  flutter_blue:
    git: git://github.com/pauldemarco/flutter_blue.git

And I am getting the ''FlutterBluePlugin.h' file not found" error in xcode when building.

dreampowder, 2020-04-04:
You can have a quick fix for this problem by recreating the ios side of the project with objective-c. 

> flutter create -i objc project_name

andrewknockin, 2020-04-05:
I am also getting the ''FlutterBluePlugin.h' file not found" error in XCode when building in Android Studio. The quick fix suggested by @dreampowder completes in the terminal, with instructions at the end to use 'flutter run', but that produces exactly the same missing file error.
While building module 'flutter_blue' imported from
    /Users/andrew/Documents/Filing/Programming/Flttr/flutter_blue/flutter_blue/i
    os/Runner/GeneratedPluginRegistrant.m:10:
    In file included from <module-includes>:1:
    /Users/andrew/Documents/Filing/Programming/Flttr/flutter_blue/flutter_blue/b
    uild/ios/Debug-iphonesimulator/flutter_blue/flutter_blue.framework/Headers/f
    lutter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not
    found
    #import "FlutterBluePlugin.h"
            ^~~~~~~~~~~~~~~~~~~~~

dreampowder, 2020-04-05:
@andrewknockin did you run the "flutter clean" command? 
this error is based on the framework support came with cocoapods + swift. it is not possible to get this error from flutter_blue.framework because there is no framework generated with objective-c.

andrewknockin, 2020-04-05:
@dreampowder yes I have just run flutter clean in the project directory. Still get the error with flutter run. It actually says 'build done' before the error.
Xcode build done.                                           24.1s
Failed to build iOS app
Error output from Xcode build:

andrewknockin, 2020-04-05:
@dreampowder I seem to have got past that error now. I had a copy of the library as another adjacent project. That contained the only copy of FlutterBluePlugin.h I could find on my system. I have totally deleted that copy of the library and run flutter clean. Now flutter run gives an error output from CocoaPods which should be in a different thread. Thanks for your help.
Automatically assigning platform `iOS` with version `10.0` on target
    `Runner` because no platform was specified. Please specify a platform for
    this target in your Podfile. See
    `https://guides.cocoapods.org/syntax/podfile.html#platform`.

dreampowder, 2020-04-06:
@andrewknockin thats not an important error, but if you really want it to go away, you can remove the. # from the version line found in the podfile file found in your ios folder

AjaCoder, 2020-04-21:
I and my team have encountered this problem several times. 
I can't say I understand it as I got the project to work one time but the problem had returned with no apparent change.
In the end, the solution that worked for us is to go to the library and set the headers as public and not project level.
If this action has side effects I'm not aware I would be happy to hear that.

[image]

Hope it helps.

woopluz, 2020-09-18:
the code insert where? thanks

yuriboeira11tx, 2021-04-07:
> Hi @ipconfiger
> I just commited a change that I confirmed working for the example app and some swift projects. Please try with the latest version of flutter by changing your pubspec to target the [...]

this works, thanks!