pauldemarco/flutter_blue #516: BluetoothCharacteristic.write(data, xxx)

DennisJu007, 2020-03-23:
The sent data is too large, how to send it in packets.

juliansteenbakker, 2020-03-23:
[Take a look at this answer.](https://stackoverflow.com/questions/24135682/android-sending-data-20-bytes-by-ble) The idea is that you send the data in separate parts, and use the first or second byte of your packet to identify which part of the total package you are sending. 

Another possibility is to change the MTU size. Changing the MTU size is really increasing your package length. This is what the readme of flutter_blue has to say about that:

> Read the MTU and request larger size
> final mtu = await device.mtu.first;
> await device.requestMtu(512);
> Note that iOS will not allow that you request the MTU size, but will always try to negotiate the highest possible MTU (iOS supports up to MTU size 185)

closed 2020-03-31 by pauldemarco

trungdd9, 2021-04-06:
> [Take a look at this answer.](https://stackoverflow.com/questions/24135682/android-sending-data-20-bytes-by-ble) The idea is that you send the data in separate parts, and use the first or second [...]

it doesn't work

> > D/BluetoothGatt(32503): configureMTU() - device: 78:A5:04:57:99:4D mtu: 512
> > D/BluetoothGatt(32503): configureMTU() - device: 78:A5:04:57:99:4D mtu: 512
> > D/BluetoothGatt(32503): onConnectionUpdated() - Device=78:A5:04:57:99:4D interval=6 latency=0 timeout=500 status=0
> > D/BluetoothGatt(32503): onConfigureMTU() - Device=78:A5:04:57:99:4D mtu=23 status=0
> > D/FlutterBluePlugin(32503): [onMtuChanged] mtu: 23 status: 0