pauldemarco/flutter_blue #518: Build Exception when build under flutter channel beta for Android

ride4sun, 2020-03-26:
I use the flutter_blue version from @boskokg 
```
flutter_blue:
    git:
      url: https://github.com/boskokg/flutter_blue.git
```

And get the following build exception when I build for Android. IOS works (at least it builds).
Any idea @boskokg @pauldemarco 

```
#0      throwToolExit (package:flutter_tools/src/base/common.dart:14:3)
#1      RunCommand.runCommand (package:flutter_tools/src/commands/run.dart:568:7)
<asynchronous suspension>
#2      FlutterCommand.verifyThenRunCommand (package:flutter_tools/src/runner/flutter_command.dart:711:18)
#3      _rootRunUnary (dart:async/zone.dart:1134:38)
#4      _CustomZone.runUnary (dart:async/zone.dart:1031:19)
#5      _FutureListener.handleValue (dart:async/future_impl.dart:140:18)
#6      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:682:45)
#7      Future._propagateToListeners (dart:async/future_impl.dart:711:32)
#8      Future._completeWithValue (dart:async/future_impl.dart:526:5)
#9      _AsyncAwaitCompleter.complete (dart:async-patch/async_patch.dart:34:15)
#10     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:293:13)
#11     RunCommand.usageValues (package:flutter_tools/src/commands/run.dart)
#12     _rootRunUnary (dart:async/zone.dart:1134:38)
#13     _CustomZone.runUnary (dart:async/zone.dart:1031:19)
#14     _FutureListener.handleValue (dart:async/future_impl.dart:140:18)
#15     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:682:45)
#16     Future._propagateToListeners (dart:async/future_impl.dart:711:32)
#17     Future._completeWithValue (dart:async/future_impl.dart:526:5)
#18     _AsyncAwaitCompleter.complete (dart:async-patch/async_patch.dart:34:15)
#19     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:293:13)
#20     AndroidDevice.isLocalEmulator (package:flutter_tools/src/android/android_device.dart)
#21     _rootRunUnary (dart:async/zone.dart:1134:38)
#22     _CustomZone.runUnary (dart:async/zone.dart:1031:19)
#23     _FutureListener.handleValue (dart:async/future_impl.dart:140:18)
#24     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:682:45)
#25     Future._propagateToListeners (dart:async/future_impl.dart:711:32)
#26     Future._completeWithValue (dart:async/future_impl.dart:526:5)
#27     Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:556:7)
#28     _rootRun (dart:async/zone.dart:1126:13)
#29     _CustomZone.run (dart:async/zone.dart:1023:19)
#30     _CustomZone.runGuarded (dart:async/zone.dart:925:7)
#31     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:965:23)
#32     _microtaskLoop (dart:async/schedule_microtask.dart:43:21)
#33     _startMicrotaskLoop (dart:async/schedule_microtask.dart:52:5)
#34     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:118:13)
#35     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:169:5)
```

fluttuer doctor
```
jay@mini obdchk % flutter doctor -v
[✓] Flutter (Channel beta, v1.15.17, on Mac OS X 10.15.3 19D76, locale en-US)
    • Flutter version 1.15.17 at /Users/jay/development/flutter
    • Framework revision 2294d75bfa (3 weeks ago), 2020-03-07 00:28:38 +0900
    • Engine revision 5aff311948
    • Dart version 2.8.0 (build 2.8.0-dev.12.0 9983424a3c)

 
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at /Users/jay/Library/Android/sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-29, build-tools 29.0.2
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_212-release-1586-b4-5784211)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 11.3.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.3.1, Build version 11C504
    • CocoaPods version 1.9.0

[✓] Android Studio (version 3.6)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 44.0.2
    • Dart plugin version 192.7761
    • Java version OpenJDK Runtime Environment (build 1.8.0_212-release-1586-b4-5784211)

[✓] VS Code (version 1.43.0)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.8.1

[✓] Connected device (3 available)
    • moto g 7     • ZY225G4LJN                               • android-arm64 • Android 9 (API 28)
    • Jay’s iPhone • bc987fccd61b08a374d7fd1329958f428155b60b • ios           • iOS 13.3.1
    • iPhone 8     • 2878EA5A-D561-47E4-969E-B231E23455C2     • ios           • com.apple.CoreSimulator.SimRuntime.iOS-13-3 (simulator)

```

boskokg, 2020-03-26:
@ride4sun I advise you to use ref in your dependency.
Can you please tell me what is the exception message?
Can you please check if it works with dev channel just to specify the problem?
I am using 1.16.2 flutter (dev) and it works...

ride4sun, 2020-03-26:
ok, will - might take me two days. I report back...

ride4sun, 2020-03-27:
Ok tried again this time with the right ref but I am still on the beta version which should not be the problem. (But I can switch to dev channel). Its a build exception not a runtime.

This is the compile error I am getting:
```
launching lib/main.dart on moto g 7 in debug mode...
Running Gradle task 'assembleDebug'...

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* Where:
Build file '/Users/jay/development/flutter/.pub-cache/git/flutter_blue-4c1d4302634af5c6049f2c7ff5a44b198422fab1/android/build.gradle' line: 37

* What went wrong:
A problem occurred evaluating project ':flutter_blue'.
> org/gradle/api/attributes/LibraryElements

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
A problem occurred configuring project ':flutter_blue'.
> compileSdkVersion is not specified.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 21s
Exception: Gradle task assembleDebug failed with exit code 1
```

I think its related to :
```
[        ] 2: Task failed with an exception.
[        ] -----------
[        ] * What went wrong:
[        ] A problem occurred configuring project ':flutter_blue'.
[        ] > compileSdkVersion is not specified.

```

boskokg, 2020-03-27:
@ride4sun I checked now starting the sample on android and it works without any problem.

I started with:

[✓] Flutter (Channel beta, v1.15.17, on Mac OS X 10.15.4 19E266, locale en-RS)
    • Flutter version 1.15.17 at /Users/boskopopovic/Desktop/work/Flutter/flutter
    • Framework revision 2294d75bfa (3 weeks ago), 2020-03-07 00:28:38 +0900
    • Engine revision 5aff311948
    • Dart version 2.8.0 (build 2.8.0-dev.12.0 9983424a3c)

Please check how the sample is configured and tell me if there is something wrong with the sample to fix it.

ride4sun, 2020-03-29:
Ok, I will - give me a day.

pauldemarco, 2020-03-31:
Starting at version 0.7.0, flutter_blue will be pinned to each stable flutter release.
This should alleviate the "moving target" problems we've seen in the past, and some of the build errors caused by the volatile project templates.

ride4sun, 2020-04-01:
It was created approx a year ago