pauldemarco/flutter_blue #521: iOS 13.4 -  Unable to scan devices

YuvalPy, 2020-03-31:
Using the following official example:

```
// Start scanning
flutterBlue.startScan(timeout: Duration(seconds: 4));

// Listen to scan results
var subscription = flutterBlue.scanResults.listen((results) {
    // do something with scan results
    for (ScanResult r in results) {
        print('${r.device.name} found! rssi: ${r.rssi}');
    }
});

// Stop scanning
flutterBlue.stopScan();
```

and building the code on my physical phone, Iphone X, version 13.4, I can't manage to detect any Bluetooth device around me. There are indeed detectable devices using light blue. There are no errors or warnings. Any idea what's the problem here?

pauldemarco, 2020-03-31:
Hi @YuvalPy,

I just tested on iPhone Xs with 13.4 with no problems.

I would inquire further on whether you have the proper permissions accepted on startup, but if you are using the example app, it would state "Bluetooth Adapter is unauthorized" like so: 
[image]

Please attempt to search local devices with nRF Connect and let me know the results.

YuvalPy, 2020-04-01:
Hi Paul,

Thank you for your fast response.

After additional testings and using the official example code, it works fine.

Thank you!

closed 2020-04-01 by YuvalPy