pauldemarco/flutter_blue #523: How to make your device act as beacon using BLE

maherjaafar, 2020-04-01:
Hi, I found how to search for **BLE** devices but I want to **make** my **android/iOS** device acting as a beacon (AltBeacon or iBEACON).

I already found a plugin named **_beacon_broadcast_** but he **didn't use BLE.**

Someone can help me please? 💯

rcauquil, 2020-04-09:
@maherjaafar any luck ?

maherjaafar, 2020-04-17:
> @maherjaafar any luck ?

I found a plugin named flutter_broadcast but I can't customize the name of the beacon :/

jjakob666, 2020-06-14:
beacon_broadcast does use BLE. I'm using it for that purpose. On android it can only broadcast AltBeacon and on IOS it can only broadcast iBeacon, so if you need to also read from these devices you have to use a library that can recognize both iBeacon and AltBeacon.  I was able to read these with flutter_beacon.

kiokumicu, 2020-06-15:
> beacon_broadcast does use BLE. I'm using it for that purpose. On android it can only broadcast AltBeacon and on IOS it can only broadcast iBeacon, so if you need to also read from these devices you have to use a library that can recognize both iBeacon and AltBeacon. I was able to read these with flutter_beacon. you can test with the appstore program BeaconScope and read beacons broadcast from beacon_broadcast.

In beacon_broadcast you can set layout and on both platforms (Android and iOS) you will have same advertising type.
But unfortunately in beacon_broadcast you can't set interval or other detailed settings (
It will be nice if we can using flutter_blue for this purpose.