pauldemarco/flutter_blue #524: BluetoothDevice name is not what I expect on iOS devices

pauldemarco, 2020-04-02:
Often times the scan results yield names that I do not expect, and cannot change when I'm using an iOS device.

pauldemarco, 2020-04-02:
From iOS CBPeripheral docs:

> A peripheral may have two different name types: one that the device advertises and another that the device publishes in its database as its Bluetooth low energy Generic Access Profile (GAP) device name. If a peripheral has both types of names, this property returns its GAP device name.

The GAP device name is prioritized over anything else.

RafaRuiz, 2020-05-05:
If anyone wants a helpful tip, I assume the Advertising name isn't going to change, and I create a map that lasts all the runtime.
Then, at the same time I scan results, I annotate this name in a map:

something like:

  extension BluetoothDeviceNameUtils on List<ScanResult> {
    void saveLocalNamesFromScanResults(List<ScanResult> scanResults) {
      scanResults.forEach((it) {
        results[it.device.id.id] = it.advertisementData.localName;
      });
    }
  }

now, what I do, if I'm interested in this name, I create an extension on the device that will tell me that name or the default if there's any error grabbing it.

  String get advertisingName => results[this.id.id] ?? this.name;

(Keep your map at runtime!)

Mohitkau, 2022-05-05:
[image]

This works for me. I have done changes in flutter_blue (src/bluetooth_device.dart) and add new method BluetoothDevice.fromProtos(protos.ScanResult p) 

[image]

then one more change in (src/flutter_blue.dart) see above screenshot.

Mohitkau, 2022-05-05:
> If anyone wants a helpful tip, I assume the Advertising name isn't going to change, and I create a map that lasts all the runtime. Then, at the same time I scan results, I annotate this name in a map:
> 
> something like:
> 
> ```
>   extension BluetoothDeviceNameUtils on List<ScanResult> {
>     void saveLocalNamesFromScanResults(List<ScanResult> scanResults) {
>       scanResults.forEach((it) {
>         results[it.device.id.id] = it.advertisementData.localName;
>       });
>     }
>   }
> ```
> 
> now, what I do, if I'm interested in this name, I create an extension on the device that will tell me that name or the default if there's any error grabbing it.
> 
> ```
>   extension BluetoothDeviceExtensions on BluetoothDevice {
>     String get advertisingName => results[this.id.id] ?? this.name;
>   }
> ```
> 
> (Keep your map at runtime!)

Thanks, for giving me the idea that's why I achieved it.