pauldemarco/flutter_blue #525: Characteristics are sending duplicate notifications when device is reconnected

pauldemarco, 2020-04-02:
Hi, there!
Still have duplicated characteristics with v0.6.3+1 and v0.7 as well.
I've spent a lot of time on this issue. I need help if you are welcome.

Android (Pixel 3a XL, 10)

> Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel dev, v1.16.1, on Mac OS X 10.15.4 19E266, locale en-UA) 
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 11.3.1)
[✓] Chrome - develop for the web
[✓] Android Studio (version 3.6)
[✓] Connected device (4 available)
• No issues found!

_**### Steps to reproduce:**_

**1)Scan for devices...**

>[AIRBOX] BluetoothState = BluetoothState.on
[AIRBOX] isScanning = true
D/BluetoothAdapter( 4005): isLeEnabled(): ON
D/BluetoothLeScanner( 4005): onScannerRegistered() - status=0 scannerId=9 mScannerId=0
D/BluetoothAdapter( 4005): isLeEnabled(): ON
[AIRBOX] isScanning = false

**2)Find target device and connect...**

>D/BluetoothGatt( 6988): connect() - device: #, auto: false
D/BluetoothGatt( 6988): registerApp()
D/BluetoothGatt( 6988): registerApp() - UUID=#
D/BluetoothGatt( 6988): onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt( 6988): onClientConnectionState() - status=0 clientIf=9 device=#
D/FlutterBluePlugin( 6988): [onConnectionStateChange] status: 0 newState: 2
[AIRBOX] STATE BluetoothDeviceState.connected
D/BluetoothGatt( 6988): discoverServices() - device: #
[AIRBOX] MTU size is 20
D/BluetoothGatt( 6988): onConnectionUpdated() - Device=# interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 6988): onSearchComplete() = Device=# Status=0
D/FlutterBluePlugin( 6988): [onServicesDiscovered] count: 3 status: 0
D/BluetoothGatt( 6988): configureMTU() - device: # mtu: 185
D/BluetoothGatt( 6988): onConnectionUpdated() - Device=# interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt( 6988): onConfigureMTU() - Device=# mtu=185 status=0
D/FlutterBluePlugin( 6988): [onMtuChanged] mtu: 185 status: 0
D/BluetoothGatt( 6988): setCharacteristicNotification() - uuid: # enable: true
D/FlutterBluePlugin( 6988): [onDescriptorWrite] uuid: # status: 0

**3)Write/read data...**

>[AIRBOX] sent: Command.mqtt_get
D/FlutterBluePlugin( 6988): [onCharacteristicWrite] uuid: # status: 0
D/FlutterBluePlugin( 6988): [onCharacteristicChanged] uuid: #
[AIRBOX] consume: Command.mqtt_get
[AIRBOX] {"status":"ok","command":"mqtt_get","payload":{"uri":"","user":"","path":"","port":0}}

**4)Disconnect...**

>D/BluetoothGatt( 6988): cancelOpen() - device: #
D/BluetoothGatt( 6988): onClientConnectionState() - status=0 clientIf=9 device=#
D/FlutterBluePlugin( 6988): [onConnectionStateChange] status: 0 newState: 0
D/BluetoothGatt( 6988): close()
D/BluetoothGatt( 6988): unregisterApp() - mClientIf=9
[AIRBOX] STATE BluetoothDeviceState.disconnected

**And when I repeat steps 2) & 3), - I have duplicated data..**

>[AIRBOX] sent: Command.mqtt_get
D/FlutterBluePlugin( 6988): [onCharacteristicWrite] uuid: # status: 0
D/FlutterBluePlugin( 6988): [onCharacteristicChanged] uuid: #
[AIRBOX] consume: Command.mqtt_get
[AIRBOX] {"status":"ok","command":"mqtt_get","payload":{"uri":"","user":"","path":"","port":0}}
[AIRBOX] {"status":"ok","command":"mqtt_get","payload":{"uri":"","user":"","path":"","port":0}}

**connect() method:**

```
void connect(BluetoothDevice device) async {
    await _device.connect(autoConnect: false);

    _stateSubscription = _device.state.listen((state) async {
      if (state == BluetoothDeviceState.disconnected) {
        await _stateSubscription.cancel();
        _stateSubscription = null;
      }

      if (state == BluetoothDeviceState.connected) {
        _mtuSubscription = _device.mtu.listen((mtu) async {
          await notifyCharacteristic.setNotifyValue(true);
        });

        _servicesSubscription = _device.services.listen((services) async {
          var service = services.first;
          notifyCharacteristic = service.characteristics.firstWhere((c) => c.uuid.toString().toUpperCase() == TX_NOTIFY_CHARACTERISTIC_UUID);
          writeCharacteristic = service.characteristics.firstWhere((c) => c.uuid.toString().toUpperCase() == RX_WRITE_CHARACTERISTIC_UUID);

          _notifySubscription = notifyCharacteristic.value.listen((value) {
            //Proceed with data...
          });
          _device.requestMtu(PREFERRED_MTU_SIZE);
        });

        _device.discoverServices();
      }
    });
  }
```

**disconnect() method**

```
Future<void> disconnect() {
    return _device.disconnect().then((value) => _clearResources());
  }

Future<void> _clearResources() async {
    await _mtuSubscription.cancel();
    _mtuSubscription = null;
    await _servicesSubscription.cancel();
    _servicesSubscription = null;
    await _notifySubscription.cancel();
    _notifySubscription = null;
    notifyChannel = null;
    writeChannel = null;
  }
```
@pauldemarco

_Originally posted by @Vizhan in https://github.com/pauldemarco/flutter_blue/issues/317#issuecomment-607268143_

Vizhan, 2020-04-02:
@pauldemarco Thnx for your time.

endrelovas, 2020-04-03:
Are you sure it is related to your framework? Are you even sure it is duplicate? Just testing on iOS I experience the following:
- receive multiple notifications (1..n)
- Restart subscription.
- you receive the notification N again. Then you receive the 1st.
- Restart whole app (swipe out from task manager).
- you receive the notification N again (!?) <= note, the app was terminated completely before
- Restart the whole device
- you won't receive duplicate

For me it seemt that the issue is that it is cached. If you just look at the characteristic value without executing the actual read command, the last "N" is still there. Once you subscribe you immediately receive that value.

Do I missunderstand something?

endrelovas, 2020-04-03:
I played around a bit in your code. Frankly: only iOS side.

I commented out line 489 FlutterBluePlugin.m removed one of notifications.
What is the exact reason this read is called?
```
  // Request a read
  //[peripheral readValueForDescriptor:cccd];
```

endrelovas, 2020-04-04:
@pauldemarco @Vizhan I had the chance to test today extensively. The above fix eliminiated all of my duplicates and residual cache problems on iIOS. Literally all.

**ROUND 1)**
> I/flutter (12242): [V]  Discovered services
> I/flutter (12242): [V]  Signing up for notifyValue: RAPC - [00002a52-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> D/BluetoothGatt(12242): setCharacteristicNotification() - uuid: 00002a52-0000-1000-8000-00805f9b34fb enable: true
> I/flutter (12242): [V]  Signed up for notifyValue: RAPC - [00002a52-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> I/flutter (12242): [D]  Signed up to RACP Services
> I/flutter (12242): [V]  Signing up for notifyValue: Glucose - [00002a18-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> D/BluetoothGatt(12242): setCharacteristicNotification() - uuid: 00002a18-0000-1000-8000-00805f9b34fb enable: true
> I/flutter (12242): [V]  Signed up for notifyValue: Glucose - [00002a18-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> I/flutter (12242): [D]  Signed up to glucose Services
> I/flutter (12242): [D]  Sending value to characteristic: [1, 1]
> I/flutter (12242): [D]  Receivied Glucose value [11, 2, 0, 228, 7, 1, 28, 6, 38, 53, 120, 0, 96, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 3, 0, 228, 7, 1, 30, 19, 31, 34, 65, 0, 89, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 4, 0, 228, 7, 2, 15, 19, 37, 20, 65, 0, 103, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 5, 0, 228, 7, 2, 21, 18, 29, 22, 65, 0, 148, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 6, 0, 228, 7, 3, 1, 8, 14, 26, 65, 0, 88, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 7, 0, 228, 7, 3, 1, 8, 55, 30, 65, 0, 175, 177, 248, 0, 0]
> I/flutter (12242): [D]  Receivied RAPC value [6, 0, 1, 1]

**ROUND 2:**
> I/flutter (12242): [D]  BLERoche: deviceState changed: BluetoothDeviceState.connected
> I/flutter (12242): [V]  Discovered services
> I/flutter (12242): [V]  Signing up for notifyValue: RAPC - [00002a52-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> D/BluetoothGatt(12242): setCharacteristicNotification() - uuid: 00002a52-0000-1000-8000-00805f9b34fb enable: true
> I/flutter (12242): [V]  Signed up for notifyValue: RAPC - [00002a52-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> I/flutter (12242): [D]  Signed up to RACP Services
> I/flutter (12242): [V]  Signing up for notifyValue: Glucose - [00002a18-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> D/BluetoothGatt(12242): setCharacteristicNotification() - uuid: 00002a18-0000-1000-8000-00805f9b34fb enable: true
> I/flutter (12242): [V]  Signed up for notifyValue: Glucose - [00002a18-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> I/flutter (12242): [D]  Signed up to glucose Services
> I/flutter (12242): [D]  Sending value to characteristic: [1, 1]
> I/flutter (12242): [D]  Receivied Glucose value [11, 2, 0, 228, 7, 1, 28, 6, 38, 53, 120, 0, 96, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 3, 0, 228, 7, 1, 30, 19, 31, 34, 65, 0, 89, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 4, 0, 228, 7, 2, 15, 19, 37, 20, 65, 0, 103, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 5, 0, 228, 7, 2, 21, 18, 29, 22, 65, 0, 148, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 6, 0, 228, 7, 3, 1, 8, 14, 26, 65, 0, 88, 176, 248, 0, 0]
> I/flutter (12242): [D]  Receivied Glucose value [11, 7, 0, 228, 7, 3, 1, 8, 55, 30, 65, 0, 175, 177, 248, 0, 0]
> I/flutter (12242): [D]  Receivied RAPC value [6, 0, 1, 1]

Vizhan, 2020-04-04:
@endrelovas I'll be able to look your solution only on Monday.

pauldemarco, 2020-04-04:
Hi @endrelovas Thanks for the excellent writeup.
I have recently published version 0.7.1 which addresses this.
I left the BehaviorSubject stream since it funnels in manual reads, but removed the virtual notifications from the write commands. Please see here: https://github.com/pauldemarco/flutter_blue/commit/487d98eb70a442749aa950256059d98d18cba91a

Would you be able to run this version through your tests and report back the results?

endrelovas, 2020-04-05:
@pauldemarco  If I understand BehaviorSubject correctly it will resend the last notification to a newly connected listener. Is this a misinterpretation?

endrelovas, 2020-04-05:
@pauldemarco while it made the thing somewhat better it is not perfect. 

**Android**
Now I receive an empty message on _every_ listen.

> I/flutter (26885): [V]  Signed up for notifyValue: Glucose - [00002a18-0000-1000-8000-00805f9b34fb] - DeviceId: [74:E1:82:0A:AB:73]
> I/flutter (26885): [D]  Signed up to glucose Services
> I/flutter (26885): [D]  Sending value to characteristic: [1, 1]
> I/flutter (26885): [D]  Receivied Glucose value []
> I/flutter (26885): [D]  Receivied Glucose value [11, 2, 0, 228, 7, 1, 28, 6, 38, 53, 120, 0, 96, 176, 248, 0, 0]
> I/flutter (26885): [D]  Receivied Glucose value [11, 3, 0, 228, 7, 1, 30, 19, 31, 34, 65, 0, 89, 176, 248, 0, 0]
> I/flutter (26885): [D]  Receivied Glucose value [11, 4, 0, 228, 7, 2, 15, 19, 37, 20, 65, 0, 103, 176, 248, 0, 0]
> I/flutter (26885): [D]  Receivied Glucose value [11, 5, 0, 228, 7, 2, 21, 18, 29, 22, 65, 0, 148, 176, 248, 0, 0]
> I/flutter (26885): [D]  Receivied Glucose value [11, 6, 0, 228, 7, 3, 1, 8, 14, 26, 65, 0, 88, 176, 248, 0, 0]
> I/flutter (26885): [D]  Receivied Glucose value [11, 7, 0, 228, 7, 3, 1, 8, 55, 30, 65, 0, 175, 177, 248, 0, 0]
> I/flutter (26885): [D]  Receivied RAPC value [6, 0, 1, 1]

**iOS**
It won't compile on iOS claiming a protobuff version issue. I don't know how to address it. Cocoapods is the most recent. Deleted the ios/Pods, ios/Podfile and ios/Podfile.lock, no luck.

> CocoaPods could not find compatible versions for pod "Protobuf"

prashant3285, 2020-04-05:
@pauldemarco v 0.7.1 doesn't solves the problem.
on re connection getting  read / write call back = number of time the device reconnected.

endrelovas, 2020-04-14:
@pauldemarco just pinging you if you had a chance to review the above?

FWeissenb, 2020-04-23:
@endrelovas had same observation with https://github.com/pauldemarco/flutter_blue/issues/378

Atm I had the problem that if I want to make a cancer()l to this characteristic value subscription it seems to not canceling both merged streams and if i listen again the callback is thrown two times. 
If i remove the _value.stream from the merge everything works like expected.

aleblsv, 2020-09-30:
Hi @FWeissenb 
I had the same problem with characteristics duplicate on Galaxy S20.
Removing  _value.stream from the Rx.merge actually not works and I still have duplicates.

endrelovas, 2020-09-30:
> Hi @FWeissenb
> I have the same problem with characteristics duplicate on Galaxy S20.
> Removing _value.stream from the Rx.merge actually not works and I still have duplicates.

Try 0.7.0 version. That is the one working for me with the AM modification. Since the change I face no issues. Paul's fixed version did not fix it.

aleblsv, 2020-09-30:
@endrelovas 0.7.2 version on Galaxy S8 working as expected. I have problem only with S20. 
Will try 0.7.0 with modification.

Chakib-Temal, 2020-11-26:
Same problem on Samsung T290 
i have found a mini solution , but i'm not that's  clean solution 
exemple with Oxymeter BLE sensor : 

```
          StreamSubscription<BluetoothDeviceState> bleDeviceStream;
          StreamSubscription<List<int>> notifyStream;

          bleDeviceStream = device.state.listen((event) async{

            if (event.toString() == "BluetoothDeviceState.disconnected"){
              await device.disconnect();
              device = null;
              bleDeviceStream.cancel();
              notifyStream.cancel();
            }

            if (event.toString() == "BluetoothDeviceState.connected"){

              if (device == null ){ // not really necessary because we already cancel the stream
                return;
              }

              List<BluetoothService> services = await device.discoverServices();
              BluetoothService SERVICE_UUID = services[3];
              RX_CHAR_UUID = SERVICE_UUID.characteristics[0];

              var _lock = new Lock(); //   synchronized: ^2.2.0+2 pubsep.yaml
              await _lock.synchronized(() async{
                await RX_CHAR_UUID.setNotifyValue(true);
              });

              notifyStream = RX_CHAR_UUID.value.listen((value) {

                if (device == null){
                  return;
                }

                print (value);

                if (value.isEmpty)
                  return;

                int BYTE0 = 0x81 & 0xFF;
                int byte0 = value[0] & 0xFF;

                if(byte0 == BYTE0) {
                  if(value[1] != 0xFF && value[2] != 0x7F && value[3] != 0x00) {
                    int sat =  value[2];
                    int fq = value[1];
                    print("heartRate= $fq oxygenSaturation= $sat");
                  }
                }
              });
            }
          });

```

aleblsv, 2020-11-26:
@Chakib-Temal In my case I solve it by total remove **await device.disconnect();** from the code.

realrk69, 2020-12-01:
Is this problem resolved? I'm having it on Android 10.0 (Mi 9T Pro) with the latest version. After reconnecting to the device or restarting the app, many times I receive duplicate notifications and in many cases, the notifications continuously start printing and I have to restart the app.

Chakib-Temal, 2020-12-01:
> Is this problem solvable? I'm having it on Android 10.0 (Mi 9T Pro) with the latest version. After reconnecting to the device or restarting the app, many times I receive duplicate notifications and in many cases, the notifications continuously start printing and I have to restart the app.

use my solution to resolve the problem 
https://github.com/pauldemarco/flutter_blue/issues/525#issuecomment-734281294

you need to listen to the deconnect event and cancel the steam (where you receive the data from ble gatt)
```bleDeviceStream.cancel();
      notifyStream.cancel();```

to check if it's a celan solution :
i used a call to a native sdk (android) via ChannelMethod 
and check with this code Java 

```
BluetoothManager manager = (BluetoothManager) getSystemService(BLUETOOTH_SERVICE); 
List<BluetoothDevice> connected = manager.getConnectedDevices(BluetoothProfile.GATT_SERVER);
Log.i(TAG, "Connected Devices:   "+  connected.size()+"");
```
there is no problem on native sdk (after multiple connect/deconnect), the problem is just on flutter party i think

endrelovas, 2020-12-01:
Just edit:

bluetooth_characteristics.dart with this.

BehaviorSubject<List<int>> _value;
Stream<List<int>> get value => Rx.merge([
      //_value.stream,
      _onValueChangedStream,
    ]);

It works without any duplication, null creation etc.

Chakib-Temal, 2020-12-01:
> Just edit:
> 
> bluetooth_characteristics.dart with this.
> 
> BehaviorSubject<List> _value;
> Stream<List> get value => Rx.merge([
> //_value.stream,
> _onValueChangedStream,
> ]);
> 
> It works without any duplication, null creation etc.

can't do this, we are a group of developpers on this project
this file is on the library, so it's need be changed by @pauldemarco if it's a really solution

realrk69, 2020-12-01:
@endrelovas Thanks, I'll give that a shot.

@Chakib-Temal You know you can download the dependency and make the changes you want right? So it will look like:
```
dependencies:
  flutter:
    sdk: flutter
  flutter_reactive_ble:
    path: ../ //Path to your locally downloaded library/dependency
```

endrelovas, 2020-12-01:
> can't do this, we are a group of developpers on this project
> this file is on the library, so it's need be changed by @pauldemarco if it's a really solution

My solution was to fork it on GitHub, commit my changes. Then link in pubspec.yaml like so:
```
  flutter_blue:
    git:
      url: git://github.com/youraccount/yourfork.git
```

realrk69, 2020-12-02:
@endrelovas Is your fork stable? Will you be maintaining it by any chance? Or should I migrate? It's already a pain due to so many changes in flutter_reactive_ble. I really like the way this library is written. Quite simple honestly. I can contribute too

spiderion, 2020-12-27:
> Just edit:
> 
> bluetooth_characteristics.dart with this.
> 
> BehaviorSubject<List> _value;
> Stream<List> get value => Rx.merge([
> //_value.stream,
> _onValueChangedStream,
> ]);
> 
> It works without any duplication, null creation etc.

looks like this solution doesn't work in my case

spiderion, 2020-12-27:
I have solved in another way.
I have added this method in the FlutterBlue class 
and I call it whenever I disconnect the device.

`  void reset() async { 
    await _methodStreamController?.close();
    _methodStreamController = StreamController.broadcast();
  }`

vaibhiarora03, 2021-02-24:
> Just edit:
> 
> bluetooth_characteristics.dart with this.
> 
> BehaviorSubject<List> _value;
> Stream<List> get value => Rx.merge([
> //_value.stream,
> _onValueChangedStream,
> ]);
> 
> It works without any duplication, null creation etc.

@endrelovas This didn't worked for me.

Using flutter_blue: ^0.7.3.

Is there any possible solution for this? The workflow of my app requires to connect BLE device for every activity and disconnect it upon the activity completion. As of now, to me it seems to never disconnect the device once connected till the app is closed by the user.

Is there really a solution for this? I can't switch to other packages as with them I can't filter out the characteristics based upon their properties.

FXschwartz, 2021-05-06:
Just hit this same exact issue on iOS. Commenting out the `_value.stream` worked for me.

Is there any chance of this getting investigated further? With a possible permanent fix?

vaibhiarora03, 2021-05-07:
@FXschwartz Earlier I was just disconnecting the device when my task got finished. But now I cancel the stream first and then disconnect the device. This worked for me.

bigboccia, 2021-05-12:
> @FXschwartz Earlier I was just disconnecting the device when my task got finished. But now I cancel the stream first and then disconnect the device. This worked for me.

This worked for me, too.
Thanks

var notifyStream;
notifyStream = characteristic.value.listen((value) {
       ...
});   

... 

notifyStream.cancel();
await device.disconnect();

MrCsabaToth, 2021-06-13:
> Hi @endrelovas Thanks for the excellent writeup.
> I have recently published version 0.7.1 which addresses this.
> I left the BehaviorSubject stream since it funnels in manual reads, but removed the virtual notifications from the write commands. Please see here: [487d98e](https://github.com/pauldemarco/flutter_blue/commit/487d98eb70a442749aa950256059d98d18cba91a)
> 
> Would you be able to run this version through your tests and report back the results?

Just adding it here: @pauldemarco's comment means that if someone would apply @endrelovas simple comment fix that would ruin manual reads? I'm using manual reads, so just warn people here to consider this.

Hitbee-dev, 2022-03-04:
I found a complete solution to this problem.
Please check the article below.
https://github.com/pauldemarco/flutter_blue/issues/1009#issuecomment-1058834600