pauldemarco/flutter_blue #527: ios debug error when podfile content exist use_frameworks!

lhx295801268, 2020-04-02:
flutter_blue.framework/Headers/flutter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not found
    #import "FlutterBluePlugin.h"
            ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    Runner/GeneratedPluginRegistrant.m:10:9: fatal error: could not build module 'flutter_blue'
    @import flutter_blue;
     ~~~~~~~^~~~~~~~~~~~

pauldemarco, 2020-04-02:
Hi @lhx295801268 I have pushed a potential fix to the master branch, if you'd like to test, just target flutter_blue master in your pubspec.yaml:
```
dependencies:
  flutter_blue:
    git: git://github.com/pauldemarco/flutter_blue.git
```

Ahmadre, 2020-07-11:
same here! Can't debug or build for ios!

Ahmadre, 2020-07-14:
Error is not disappearing after setting that path on pubspec